bogde/HX711 #11: issue reading A0 on display shield if HX711 loaded

mkocan, 2015-01-26:
Hi,your HX711 library was very helpful to me but currently I am facing a following problem:

I have an arduino uno with a display shield 1602 connected to it. Additionally I have an HX711 A/D connected over display shield pins in a way that A1 is connected to CK/TX and A3 is connected to DO/RX. If I try to read the A0 (meaning the keys on display shield) with analogRead(A0) everything reads ok until I try to initialize the HX711 library (with HX711 scale(A3, A1); line ) once this is added to my code the analogRead(A0) always returns number 7.

The example code that can be used to replicate is like 

#include <HX711.h>
#include <LiquidCrystal.h>

LiquidCrystal lcd(8, 9, 4, 5, 6, 7);
HX711 scale(A3, A1);	

void setup() {
  // set up the LCD's number of columns and rows:

  lcd.begin(16, 2);
  // Print a message to the LCD.
  lcd.setCursor(0,0);
  lcd.print("LCD Key Shield");
  lcd.setCursor(0,1);
  lcd.print("Press Key:");
}

void loop() {
  int x;
  x = analogRead (0);
  lcd.setCursor(10,1);
  if (x < 60) {
    lcd.print ("Right ");
  }
  else if (x < 200) {
    lcd.print ("Up    ");
  }
  else if (x < 400){
    lcd.print ("Down  ");
  }
  else if (x < 600){
    lcd.print ("Left  ");
  }
  else if (x < 800){
    lcd.print ("Select");
  }
}

If the line with HX711 scale(A3, A1); is removed then the keys are read correctly. But once I add that line I always get 'Press Key: Right' message on a display (since analogRead(A0) is always 7).
The HX711 library itself works correctly, returns values from loadcell plugged as expected, but the A0 pin is not readable in that case.

Is there any restriction for usage of other analog pins if some of them are used by HX711 lib? Am I doing anything wrong in relation to initializing the library?

  Thanks for any help,

             Marek

bogde, 2015-01-31:
i'm really sorry but due to the lack of time i can't provide any kind of support for this project anymore. also, i haven't used the chip in months and don't plan to use it soon either, i don't have an arduino uno, nor the type of display you're using. you should find a better supported library, or try to write your own. you could also try to re-route the analog input pin from your display shield, or at least use a1, a2 for the hx711 instead of a1,a3. again, i'm sorry i can't help you out. good luck.

closed 2015-01-31 by bogde

electrokean, 2015-01-31:
@mkocan This library should have no impact on other pins.
Maybe you actually have an electrical short between A0 and A1?

When the libaray is not initialised, A1 would be a high-impedance input and not affect readings of A0.  But as soon as the library is initialised, A1 gets configured as an output to drive SCK, and that would force A1 (and in turn A0) to be LOW after each read of the scale.