bogde/HX711 #12: using channel A + B together?  version 2

PacoRaap, 2015-01-31:
Hi,

Although you already had this question before your answer was not clear to me and I am still wondering if reading reading A and B channel can be done.
Your prevoius answer put some dust around it (although I understadn it can be done) and gave more questions then answers.
I try to understand your thinking but I can not, so are may others but they do not ask again or do not want to share the SOLUTION :-)

If I understand correct this code should work using your lib which I made more simple to understand what is going on in reading the raw values.
-----------------------------------------------
#include "HX711.h"
// HX711.DOUT	- pin #A1
// HX711.PD_SCK	- pin #A0
HX711 scale(A1, A0);// parameter "gain" is ommited; the default value 128 is used by the library

void setup() 
{
  Serial.begin(9600);
}
void loop()
{
 //set the gain factor; takes effect only after a call to read()
(scale.read());	// print a raw reading
set_gain(32); //set gain and channel
Serial.println(scale.read());	// print a raw reading from the ADC channel B to display
(scale.read());	// print a raw reading
set_gain(64); //set gain and channel
Serial.println(scale.read());	// print a raw reading from the ADC channel A to display
}
------------------------------------------------
Although this is what I think you tried to explain it is not working.
i get an error on set_gain(32) to start with.
I know manual says the 26 pulses to get B 32 and 27 pulses to get A 64 but does the lib support the 27 pulses to be read and convert to a raw value or only the A 128?

Your help would be highly appriciated.

Paco

bogde, 2015-01-31:
i'm really sorry but due to the lack of time i can't provide any kind of support for this project anymore. also, i haven't used the chip in months and don't plan to use it soon either. as i already explained, the channel B can be read, just like channel A, you just need to select it.

your code should read: 
```javascript
scale.set_gain(32);
```
instead of:
```javascript
set_gain(32);
```
if you already tried that, i have no other ideas. good luck!

closed 2015-01-31 by bogde

PacoRaap, 2015-01-31:
OK understand you position.

 

I reworked it to this simple sample code and it worked.

 

#include "HX711.h"

 

// HX711.DOUT                - pin #A1

// HX711.PD_SCK            - pin #A0

 

HX711 scale(A1, A0);                     // parameter "gain" is ommited; the default value 128 is used by the library

 

void setup() 

{

  Serial.begin(9600);

}

 

void loop()

{

scale.set_gain(128);

Serial.print("Loadcell A > ");

Serial.print((scale.read()));         // print a raw reading from the ADC channel B

scale.set_gain(32);

Serial.print("  Loadcell B > ");

Serial.println((scale.read()));     // print a raw reading from the ADC channel A

}

 

 

 

 

Cheers and thanks,

 

Paco

 

Met vriendelijke groeten/with kind regards/ mit freundlichen Grüssen,

 

Paco Raap

 

CUSTOM RACING DECAL

 

Custom Racing Decal is onderdeel van Jupa Creations.
Custom Racing Decal is part of Jupa Creations.

 

U kunt ons ook  vinden op Facebook/You can find us at Facebook

https://www.facebook.com/CustomRacingDecal

kozoke, 2016-06-06:
Hi Paco - 

Did you get this to work?  I tried it tonight and I only get a raw value but it does not change based on putting pressure on the gauge.  Did you truly see a fluctuating value when measuring both channels?
Thanks.
PS- Bogde - great work on this. Apologies for hijacking the thread. :-)

bogde, 2016-06-06:
Are you sure your wiring is correct? Can you post your wiring and your code?

PacoRaap, 2016-06-06:
> Are you sure your wiring is correct? Can you post your wiring and your
> code?
>
> ---
> https://github.com/bogde/HX711/issues/12#issuecomment-223885933

Hi both,
After the advice from Bogde, I put some code together but never got it to
work.
Sequence if I remember correct:
Set channel A by 128.
Read data channel A and put it in variable.
Set channel B by 32.
Read data channel B and put it in a variable.
It is already a long time ago I tried this and have to dig deeply for the
code if I am able to find it.
In meantime I use 4 x HX711 boards to read 4 loadcells, although 2 would
be nice to reduce hardware and wires.

http://forum.arduino.cc/index.php?topic=226314.105
#108
If a piece of sample code is available I will do some test.
Currently too many other project with higher priority.

Cheers, Paco

kozoke, 2016-06-07:
Hi,
1) Wiring: See wiring diagram below. In my case, red are the signal (I have channel a working fine, responding well to changes in the load cell. B just seems to wander a little but won't respond to load cell stress.)

[image]

2) Code: I used Paco's code above - included here for convenience. 

Thoughts?  Thanks

//Start code
#include "HX711.h"
// HX711.DOUT                - pin #A1
// HX711.PD_SCK            - pin #A0

HX711 scale(A1, A0);                     // parameter "gain" is ommited; the default value 128 is used by the library

void setup()
{
  Serial.begin(9600);
}

void loop()
{
scale.set_gain(128);
Serial.print("Loadcell A > ");
Serial.print((scale.read()));         // print a raw reading from the ADC channel B

scale.set_gain(32);
Serial.print("  Loadcell B > ");
Serial.println((scale.read()));     // print a raw reading from the ADC channel A
}

electrokean, 2016-06-07:
@kozoke actually, using thicker wire will reduce resistance, and lead to less error in excitation voltage.
What happens if you use `scale.read_average()` instead of `scale.read()` ?

bogde, 2016-06-07:
@kozoke (crazy thought) did you try switching the load cells (move the one currently on channel A to channel B and the other way around)?

kozoke, 2016-06-08:
@bogde - good question- I did switch them tonight. No difference. Channel B is still not responding. I also re-soldered the connections. No difference.

@electrokean - Thanks on the wire. That makes sense. I tried the read_average, no real difference. see graphs.  One with read and one with read_average.  channel A responses but B just seems to wander regardless of pressure applied to the b sensors.

Maybe I just have a bad hx711 channel B?  Any other thoughts/ideas?  I'll pull out another hx711 and give that a go this week.

Sample after re-wiring and changing my A and B channels.
[image]

Sample after changing it to read_average
[image]

Photo of overall connections
[image]
[image]
.

PacoRaap, 2016-06-08:
> ![hx711ch b

Maybe we have to go back to the fact so far if you search the web there is
no one claiming and showing that regardless the library they used
succeeded in a working setup and shared the code.
All we have seen passing by are assumptions and propositions how it should
work.
So logical thinking.
Kozoke, for your setup if it works for A channel. If you switch the wires
that currently measure B to A and the two cells are working then the cells
are OK. I can’t imagine the B channel is defective.
So getting back to the code.
Start Loop
Set channel A by 128 and measure and put it in a variable that is serial
printed.
Set channel B by 32 and measure and put it in a variable that is serial
printed.
Time out for 1 second
End Loop
If that isn’t working then what will? 
What if you only use a basic code to read the B channel?
Paco.

electrokean, 2016-06-08:
Yeah, I was going to suggest simplifying the code to just read Channel B (set gain to 32 at start, and don't change it) and see what you get.
Also, what kind of variance do you get on Channel A with weight?  You'll only see a quarter of that on Channel B.  If the load cells are quite high in weight rating, they would require extra force with the lower gain.

kozoke, 2016-06-09:
Thanks for the feedback.  Good point on the low gain and variance difference due to gain.  At this point, i'm going to shift to having two HX711s. I'll find some simple code where I can just read two separate channel A values and be done with it.  Thank you both for your time and feedback.

electrokean, 2016-06-09:
@kozoke see https://github.com/bogde/HX711/issues/35 for an example way to use multple HX711 objects

PacoRaap, 2016-06-10:
I think no one ever will have a working HX711 A+B loadcell….;-)

 

Met vriendelijke groeten/with kind regards,

 

Paco Raap

 

CUSTOM RACING DECAL

 

Custom Racing Decal is onderdeel van Jupa Creations.
Custom Racing Decal is part of Jupa Creations.

Facebook: CustomRacingDecal