bogde/HX711 #13: Increasing results over time

sher1111, 2015-09-05:
Hi,

Very well job done on library. Made my work very easy. I am building a scale with 350 kg load cell. I sucessfully got everything working. Found float number for the load cell. When I start my arduino project, everything starts just great. I put weight and the programm shows correct numbers. But when I leave item on the scale for some time (e.g. for 4 hours) the numbers start to change. They may increase or decrease over time. Changes are not very crucial, like 200-300 grams. But I am guessing it will change significantly over long time period.

Any help would be appreciated. Thanks in advance.

bogde, 2015-09-14:
hello, sorry for the delay, i just now noticed your message. unfortunately i didn't play with this in some time, and can't really help you much. the code does the same thing every time and i doubt it's a software issue, since it seems to give accurate measurements. did you check your mechanical part? maybe the frame bends, maybe some kind of deformation takes place. did you try with a smaller weight to see if you still get the same amount of error? 300 grams seems huge to me.

chaeplin, 2015-09-14:
check this http://i.imgur.com/ltYKyAG.png
Without tare(), measurements is affected by temperature over long time.

sher1111, 2015-09-23:
Thank you for your reply, bodge. You were right I had some connection problems. I did not soldered wires. 

chaeplin, I do not see the scales weight graph at the image. Is it the first graph?

chaeplin, 2015-09-23:
@sher1111 first graph.
one more  : http://i.imgur.com/uqNV7vo.png

CFTechno, 2015-09-23:
It might be the "creep" behaviour of the cells. Normally in the datasheets it will mention someting like Creep (1min)  0.05%FS.

sher1111, 2015-09-23:
Thanks a lot to you all for your help. I am planning on building scales, on which I am going to leave a weight for a long time, like 1-2 months. And check the weight change in everyday, by gprs sheild. So, all this information is important to me.

chaeplin, 2015-09-23:
@sher1111 I think this link will help https://hackaday.io/project/1741-honeybee-hive-monitoring

sher1111, 2015-09-23:
thanks for the article about bee scales, @chaeplin! That's exactly what I am building. I am a beekeeper! And I've already read the first one, but the second seems very interesting. Thanks!

closed 2015-10-30 by bogde