bogde/HX711 #15: Values "rolling over"?

jwhendy, 2015-09-25:
Very nice library! I just purchased a Futek LSM400 for use at work and purchased an HX711 pre-wired breakout board to read it with an Arduino. The datasheet for the load cell specifies an excitation voltage up to 18V:
- http://www.futek.com/files/pdf/Product%20Drawings/lsm400.pdf

I saw the specification to connect the HX711 to 5V on this site and at [SparkFun's hookup guide](https://learn.sparkfun.com/tutorials/load-cell-amplifier-hx711-breakout-hookup-guide), but thought that was for convenience (since one has 5V available from the Arduino).

My primary question with the library is that my load cell has a range of 5lbs, and I was simply stacking some plastic wheels on it from an erector set I had handy (couldn't have been more than a few hundred grams). My values started around -29k, went past zero up to +30k but then would "roll over" back to negative. What might be the symptom of this? I tried setting the scaling to 64 and 32 but got similar results with 64 and pretty much no change with 32 (values hung around 2k no matter what I did).

Unfortunately, I thought I was perhaps not giving my load cell a high enough excitation voltage so I connected a 12V power supply to the HX711 (12V+ to VCC and the - terminal to both the HX711 and Arduino GND to make them common). I got nothing in the serial monitor, and now I can't seem to get any readings from the HX711 anymore :( I'm wondering if I fried it.

In any case, my primary question is on the rollover values. My load cell is 1000k nominal resistance and 2mV/V output. The HX711 should be compatible with this, right?

If you happen to know if it's not voltage protected and could easily tell me if it's likely fried, that would be appreciated too as I'll order another.

Thanks!

bogde, 2015-09-27:
No ideas, sorry. About the board being defective, if you used to be able to read something out of it and now you aren't anymore, with the same setup, then I would just order a new one and be more careful with it.

jwhendy, 2015-09-27:
Sorry and closing. A colleague caught that I was storing the read came as an `int`... Using analog pins had me reflexively thinking 0-1023, which is obviously wrong.

Sorry for the noise and thanks for the quick response!

closed 2015-09-27 by jwhendy