bogde/HX711 #18: Turnigy Thrust Stand

alduxvm, 2015-11-30:
Good day!

I'm reading this thrust stand (http://www.hobbyking.com/hobbyking/store/__66623__Turnigy_Thrust_Stand.html) Its connected to and Arduino pro (as instructions from sparkfun HX711 site), and I'm getting some data, so, I proceed to do the calibration, I'm uploading this code to calibrate: 

```
  scale.set_scale();                 
  scale.tare();	
  delay(5000); // just to have time to put the weight on it...
  Serial.println(scale.get_units(10));
```

And I'm getting a number like `-754935.00`, my weight is 200 grams, so, I divide by that, and then I upload a code similar to this:
```
void setup() {
  Serial.begin(115200);
  scale.set_scale(-3774.675); //   -754935.00/200               
  scale.tare();	
}

void loop() {
  Serial.print("one reading:\t");
  Serial.println(scale.get_units(), 1);
  delay(1000);
}
```

And I'm getting this results with no weight:
```
one reading:	222.2
one reading:	222.2
one reading:	222.2
one reading:	222.2
one reading:	222.2
```
And this when I put my 200 grams weight:
```
one reading:	200.1
one reading:	200.1
one reading:	200.1
one reading:	200.0
one reading:	200.0
```
And this when I put my 500 grams weight:
```
one reading:	188.3
one reading:	166.3
one reading:	166.6
one reading:	166.6
one reading:	166.6
one reading:	166.6
```

What am I doing wrong? Any help?? 

thanks a lot!

bogde, 2015-11-30:
hey there,

since you're getting a negative number, you probably need to switch the S+, S- wires (or E+, E- on some boards) the ones that come from your load cell into your HX711 board. I would try that first...

zfields, 2015-11-30:
Hi @alduxvm! Would you mind testing my pull-request to see if it fixes the way your application is behaving? I have a hunch that their is a bug in the datasheet about decoding the value returned by the hx711. If you could test this, it would really clear up some confusion for both @bogde and myself.

Thanks!
Zak

alduxvm, 2015-12-03:
I'm going to try both next week, and let you know, thanks!

cybermat, 2015-12-06:
I wrote an Arduino sketch to make sure I understood the representation of 32-bit and 24-bit two's complement values.
There may be an error in the HX711.cpp code: it is toggling the sign-bit in the 24-bit value returned from the HX711 but it should be extending that sign bit into the top 8 bits of the 32-bit value returned to library user calling programs.
Attached is a copy of the test Arduino sketch with the output appended.
[HX711_adc_conversion.txt](https://github.com/bogde/HX711/files/53047/HX711_adc_conversion.txt)

I changed my version of HX711.cpp from - 

<code>	data[2] ^= 0x80;</code>
<code>	return ((uint32_t) data[2] << 16) | ((uint32_t) data[1] << 8) | (uint32_t) data[0];</code>
to -
<code></code>
<code>byte datasign;</code>
<code>	// 1 Dec 2015 - add most significant byte "datasign" to extend sign bit to 32-bit 2's complement result (from 24-bit source data from HX711 ADC)</code>
<code>	if ((data[2] & 0x80) == 0x00)</code>
<code>		datasign = 0x00;</code>
<code>	else</code>
<code>		datasign = 0xFF;</code>
<code></code>
<code>	return ((uint32_t)datasign << 24) | ((uint32_t)data[2] << 16) | ((uint32_t)data[1] << 8) | (uint32_t)data[0];</code>

zfields, 2015-12-06:
@cybermat I ran into this as well. I have a pull-request outstanding that extends the sign-bit and converts the number from two's compliment. I see you caught the padding, but you aren't converting the number from two's compliment in your pasted code. Would you mind giving it a try? Thanks!

cybermat, 2015-12-07:
@zfields, The HX711 library is fine just by extending the sign bit of each 24-bit value from the ADC into the top 8 bits of 32-bit Long (signed) integer values it produces.

The answer might be either a positive or a negative integer value (which are the two's complement of each other), and which value is correct will depend on the user's application and how they connect the HX711 ADC to their sensor. 

To change the sign of a Long integer value the user can either multiply the answer by -1 or swap the leads connecting the sensor to the HX711 ADC.

I'm using the HX711 with an air pressure sensor where negative numbers represent partial vacuums and positive numbers represent pressures above atmospheric pressure, so the HX711 library can't make any assumptions about whether the user always wants a positive value returned.

"Two's complement" has 2 different meanings. 
The first meaning is about converting positive numbers to negative numbers and vice versa. If your computer uses 2's complement notation then multiplying signed integers by -1 does this conversion (that is, the 2's complement of +3 is -3. The 2's complement of -3 is +3). 

The second meaning is that the Arduino Uno uses this way of representing Long integers, so there's no need to convert two's complement values from the HX711 ADC to some other form, such as "one's complement" notation. They only need to be made 8 bits longer (that is, the Arduino Uno represents signed Long integer values in 32 bits stored in 2's complement notation.)

zfields, 2015-12-08:
@cybermat Two's compliment only has one meaning, and I think you are over simplifying it. Your math (* -1) is true, but the bits behind your math are radically different. The data sheet reports it returns a two's compliment value. I'll use 3 bits to illustrate my observation...
```
 b  u  s |2's u  s 
000 0  0 |000 0  0
001 1  1 |111 7 -1
010 2  2 |110 6 -2
011 3  3 |101 5 -3
100 4 -4 |100 4 -4
101 5 -3 |011 3  3
110 6 -2 |010 2  2
111 7 -1 |001 1  1
```
The padding is important to preserve in the value you receive (as we have both noted). However, if it is truly returning a two's compliment value, then the result can be radically different depending whether or not the value is signed or unsigned. In this case the number is signed, so the value appears to be negative (which is flipped by the XOR with 0x80). The logic fails for the negative max case, which is wrong in an absolute sense.

zfields, 2015-12-13:
@alduxvm I wanted to send you a ping to see if you had an opportunity to test my pull request?

alduxvm, 2015-12-14:
Ok... I tried changing E+ and E-... The thrust stand stopped working.

So, I soldered again and give it a go to @zfields version of the library and tare it using this code in setup and nothing on loop:
```
  scale.set_scale();                 
  scale.tare(); 
  delay(5000); // just to have time to put the weight on it...
  Serial.println(scale.get_units(10));
```

I got a value of  -16545.00 with a weight of 40 grams... then I tested this code:
```
void setup() {
  Serial.begin(115200);
  scale.set_scale(-413.625); //   -16545.00/40               
  scale.tare(); 
}

void loop() {
  Serial.print("one reading:\t");
  Serial.println(scale.get_units(), 1);
  delay(1000);
}
```

And if I used the same 40 grams weight, I got this results:
```
one reading:	-0.1
one reading:	0.0
one reading:	28.2
one reading:	39.7
one reading:	39.8
one reading:	39.6
one reading:	39.7
one reading:	39.9
one reading:	39.9
one reading:	40.1
one reading:	40.1
one reading:	40.2
one reading:	40.4
one reading:	40.6
one reading:	40.5
one reading:	40.9
one reading:	40.6
one reading:	40.8
one reading:	41.0
one reading:	40.9
one reading:	41.0
one reading:	41.1
one reading:	41.2
```

If I put another load like 200grams, I got this result:
```
one reading:	-0.0
one reading:	0.0
one reading:	0.1
one reading:	198.3
one reading:	196.1
one reading:	196.1
one reading:	195.9
one reading:	196.0
one reading:	196.2
one reading:	196.0
```

So, I think @zfields version works!!! what do you think???

bogde, 2015-12-16:
hello all, due to the lack of time i've been unable to look into this lately. so, unless anyone has a different opinion, i will merge @zfields changes.

chaeplin, 2015-12-16:
@bogde @zfields I am using green HX711 and 4cell sensor ( body scale )
1. hx711 http://www.ebay.com/itm/Weighing-Sensor-AD-Module-Dual-Channel-24-Bit-A-D-Conversion-HX711-Shieding-HPT-/201485738669?hash=item2ee97c62ad:g:jQoAAOSwHPlWcFU9
2. cell http://www.ebay.com/itm/4-x-Body-Load-Scale-Weighing-Sensor-Resistance-Strain-50kg-Half-bridge-YZC-161B-/121685678462?hash=item1c5508197e:g:9kEAAOSwLVZVkU8V

Here is my test result. Ignore time third one and few gram of variation.
with   scale.set_scale(23040.f);  scale.tare();
Same value with -.

Current ---------------------
Time(sec) measured
8.21	: 2002	:	2002
8.66	: 2011	:	2011
9.21	: 2004	:	2004
9.66	: 2000	:	2000
10.21	: 2003	:	2003
10.66	: 2005	:	2005
11.21	: 2000	:	2000
11.66	: 2003	:	2003
12.21	: 2000	:	2000
12.67	: 2000	:	2000
13.21	: 2000	:	2000
13.66	: 2003	:	2003
14.21	: 2000	:	2000
14.67	: 2003	:	2003
15.21	: 1999	:	1999
15.67	: 2001	:	2001

With https://github.com/zfields/HX711
-----------------------------------
14.66	: -2022	:	-2022
15.21	: -1996	:	-1996
15.67	: -1992	:	-1992
16.21	: -1993	:	-1993
16.67	: -1990	:	-1990
17.21	: -1992	:	-1992
17.67	: -1992	:	-1992
18.21	: -1988	:	-1988
18.67	: -2062	:	-2062
19.21	: -2016	:	-2016
19.67	: -2020	:	-2020
20.21	: -1999	:	-1999
20.67	: -1998	:	-1998
21.21	: -1998	:	-1998
21.67	: -1998	:	-1998
22.21	: -1996	:	-1996
22.67	: -1997	:	-1997
23.22	: -1999	:	-1999
23.67	: -1999	:	-1999
24.22	: -1999	:	-1999
24.67	: -2000	:	-2000

bogde, 2015-12-16:
thanks for your feedback. can you switch wiring (basically A- with A) and test zfields version again?

zfields, 2015-12-16:
Also, are you recalibrating the scale when you put the different firmware. I would be surprised if the scale factor was the same.

chaeplin, 2015-12-16:
With switching wiring, current sketch gives - value
With switching wiring, new sketch,  scale scale.set_scale(23.f);  Serial.println(scale.get_units(5));

--- 2117g is on
2123.13
2122.91
2123.78
2122.00
2123.70

-- nothing on it < -- sudden spike
-7.30
-8.26
-6.61
-7.30
-8.04
-24671.65
-7.70
-7.57
-7.65
-7.13
-7.48
-7.96
-7.78
-6.87
-2578.70
-7.04
-6.26
-7.91
-6.91
-8.43
-7.13
-6.65
-79379.35
-7.00
-5.96
-7.61
-7.35
-6.09
-7.13
-7.04
-7.65
-6.22
-7.17
-8.43
-6.61
-7.70
-6.57
-6.22
-6.22
15230.74
-6.30
-6.74
-4.57
-5.39
-3.52
48276.91
-2.87
-7.52
2123.13 < --- place a thing
2122.91
2123.78
2122.00
2123.70
-1.35

chaeplin, 2015-12-16:
Looks like recalibrating is not needed.
scale.set_scale(23.f); gives gram unit. --> 2117
scale.set_scale(231.f); gives 100 gram unit. --> 211.7
scale.set_scale(2300.f); gives 100 gram unit. --> 21.17

chaeplin, 2015-12-17:
I have a test  again, with my original sketch and zfields's.

2.66	: -3	:	-3
3.20	: 0	:	0
3.66	: -7	:	-7
4.20	: 3	:	3
4.66	: 5	:	5
5.20	: 11	:	11
5.66	: -5	:	-5
6.20	: -23	:	-23
6.66	: 19	:	19
7.20	: 6	:	6
7.66	: -5	:	-5
8.20	: -3	:	-3
8.66	: -7	:	-7
9.20	: -2	:	-2
9.66	: 3	:	3
10.20	: 2	:	2
10.66	: -14	:	-14
11.20	: 134	:	134
11.66	: 568	:	568
12.20	: 807	:	807
12.66	: 1126	:	1126
13.20	: 1648	:	1648
13.66	: 1983	:	1983
14.20	: 2014	:	2014
14.66	: 1990	:	1990
15.20	: 1991	:	1991
15.66	: 1989	:	1989
16.20	: 1987	:	1987
16.66	: 1987	:	1987
17.20	: 1988	:	1988
17.66	: 1980	:	1980
18.20	: 1952	:	1952
18.66	: 2042	:	2042
19.20	: 2109	:	2109
19.66	: 2260	:	2260
20.20	: 2182	:	2182
20.66	: 2012	:	2012
21.21	: 2041	:	2041
21.66	: 2000	:	2000
22.20	: 1997	:	1997
22.66	: 1988	:	1988
23.20	: 1988	:	1988
23.66	: 1989	:	1989
24.21	: 1981	:	1981
24.66	: 1984	:	1984
25.21	: 2002	:	2002
25.66	: 1984	:	1984
26.20	: 1954	:	1954
26.66	: 1999	:	1999
27.21	: 2042	:	2042
27.66	: 1566	:	1566
28.21	: 114	:	114
28.66	: 5	:	5
29.20	: 5	:	5
29.66	: -11	:	-11
30.21	: -9	:	-9
30.66	: -3	:	-3
31.21	: -3	:	-3
31.66	: -2	:	-2
32.21	: -7	:	-7
32.66	: -12	:	-12
33.20	: 1	:	1
33.66	: 5	:	5
34.21	: 4	:	4
34.66	: -4	:	-4
35.21	: -2	:	-2
35.66	: -1	:	-1
36.20	: 0	:	0
36.66	: 0	:	0
37.21	: 2	:	2
37.66	: 0	:	0
38.21	: -1	:	-1
38.66	: 0	:	0
39.21	: -2	:	-2
39.66	: 1	:	1
40.21	: 1	:	1
40.66	: 0	:	0
41.21	: 1	:	1
41.66	: 3	:	3
42.21	: 0	:	0
42.66	: 0	:	0
43.21	: 0	:	0
43.66	: 0	:	0
44.21	: 4	:	4
44.66	: 0	:	0
45.21	: 1	:	1
45.66	: 0	:	0
46.21	: 0	:	0
46.66	: 0	:	0
47.21	: 0	:	0
47.66	: 0	:	0
48.21	: -10	:	-10
48.66	: -51	:	-51
49.21	: 4	:	4
49.66	: 381	:	381
50.21	: 113	:	113

zfields, 2015-12-17:
@chaeplin I don't quite understand your values (x:y:z). x = time, y = ?, z = ? I am assuming that z is the value returned from my modification. What is y? (or vice versa) I don't understand the test you are running. Are you testing both versions simultaneously?
Are you adjusting the weight on the scale? I think this fix will be observed by testing a large range from negative (or tension) weight through to positive (or compression) weight.

chaeplin, 2015-12-17:
@zfields time / weight /  ( a function not used any more) return of weight

My scale is for dog's poop pad, so no negative test.

scale
https://raw.githubusercontent.com/chaeplin/esp8266_and_arduino/master/_03-hx711-scale/pics/2.jpg

result with your mod.
https://raw.githubusercontent.com/chaeplin/esp8266_and_arduino/master/_03-hx711-scale/pics/10.png

zfields, 2015-12-18:
@chaeplin Interesting test and a pretty app!

I have a couple of questions though.
    1. Are you able to test the scale with a known weight, across a broad range (I think the scale is rated for 50kg)? You're dog's poop is not a known value, and will be roughly the same time after time, so it doesn't make for the best test.
    2. Does the noise on the scale only happen with my change set, or is it an issue with the hx711 library in general?

Thanks again for all the testing!

chaeplin, 2015-12-18:
@zfields The graph is output of grafana(db : influxdb, front : grafa) :).

1) tested with two known, 2117 and 2000.
2) both.

I use standard deviation < 15 (10 check) for reporting, so the noise is not a problem.

zfields, 2015-12-18:
@chaeplin As a final test, would you put a significant load on the scales and see how the output behaves between the two libraries?

chaeplin, 2015-12-19:
@zfields done
cur and new : https://raw.githubusercontent.com/chaeplin/esp8266_and_arduino/master/_03-hx711-scale/for_test/pics/all.png

cur : https://raw.githubusercontent.com/chaeplin/esp8266_and_arduino/master/_03-hx711-scale/for_test/pics/cur.png

new : https://raw.githubusercontent.com/chaeplin/esp8266_and_arduino/master/_03-hx711-scale/for_test/pics/new.png

closed 2015-12-25 by bogde