bogde/HX711 #20: Calculating the proper SCALE value

Bartimaeus-, 2015-12-14:
I sat down and worked out the math for actually calculating what the SCALE correction should be for a given system. Someone else can go over and check this if they want, I solved it in the formulaic form and verified it by calculating with real values and checking the results.

scale = (output*gain*adc_range*excitation)/(cell_range*vcc)

where:
output = the output rating of the load cell (mV/V). Should be in volts, if your final result looks wrong see if it is off by some factor of 10 to see if you forgot to account for the unit changes.
gain = the gain setting of the amplifier
adc_range = the range of values that the adc has (precision). For this 24 bit adc this is (2^24)-1= 16777215
excitation = the input voltage for the load cell. For Sparkfun's breakout board this is equal to vcc and the two values cancel out.
cell_range = the max rating for the load cell in whichever units you are using. I was using a load cell rated for 50KG, so to get an output in kilograms I would use the value 50 here.
vcc = the system voltage for the amplifier

I was able to get good readings on my load cell using this formula without having to guess and check.

bogde, 2015-12-16:
thanks for that. i have a green chinese board and a dfrobot one. i'll try to apply your math to see how it goes.

vicatcu, 2016-12-06:
@Bartimaeus- I'm trying to understand this and at the end of the day SCALE should have units of ADC counts per unit weight (e.g. ticks / kg). The term that is throwing me off the most is:

> output = the output rating of the load cell (mV/V). Should be in volts, if your final result looks wrong see if it is off by some factor of 10 to see if you forgot to account for the unit changes.

... [this](http://www.phidgets.com/docs/Load_Cell_Primer) primer describes output rating as:

> the difference in the output of the sensor between when it is fully loaded to its rated capacity, and when it’s unloaded. Effectively, it’s how sensitive the sensor is, and corresponds to the gain calculated when calibrating the sensor. More expensive sensors have an exact rated output based on an individual calibration done at the factory. 

So what would you use for the "output rating" of [this](http://www.phidgets.com/products.php?category=34&product_id=3134_0) load cell? Something tells me 1.0 (listed as Rated Output) isn't the right idea. If I try and reconstruct how you ended up where you ended up, I think it's predicated on the fact that you have a coefficient of linearity from the manufacturer, encapsulated in the 'output' constant (or have otherwise characterized this for yourself), right?

If I apply dimensional analysis to the equation you came up with, and 'ouput' has units of V, then SCALE ends up in units of: (V * 1 * ticks * V)/(kg * V) = V * ticks / kg... what am I missing?

Draeeannae, 2017-01-29:
Hello All!

Currently looking to use a HX711 in a school project, the force sensor we are using is a FSS020WNSB (link listed below)
https://sensing.honeywell.com/index.php?ci_id=50137

We are attempting to calculate the SCALE value for the output, but are having some difficulties. We are not finding the output rating of our load cell, which should be clearly stated within out datasheet. The sensitivity is provided, and we have been multiplying this value by the maximum load (20 N) in order to get a value with the correct units (1.8 mV/V/N * 20 N = 36 mV/V).

Additionally, looking at the discussion above, we are not sure how the adc_range accounts into the function. When we attempt to multiply this value in our calculations, it seems to inflate out output to an unrealisticly high value. Does the output value compensate for this?

Using the guideline listed above, this is where we are at:

Output: not sure how to calculate this value

adc_range: Same as provided in example above

Excitation: 5V, same as Vcc

cell_range: 20 N (from datasheet listed above)

Vcc: 5V, same as Excitation

---------------------
scale = (output * 128 * 16777215 * 5) / (20 N * 5)

Any assistance in this would be much appreciated! Thank you for your time!

bogde, 2017-01-30:
can you test and verify your math using a known weight or something like that?

i think the typical output rating for your load cell is 1.8 (sensitivity - in your datasheet). see @vicatcu 's comment above.

scale = 1 provided in the code has no special meaning, it doesn't mean your value must be close to that.

Bartimaeus-, 2017-02-02:
@vicatcu from Wikipedia:

vainell, 2017-06-11:
guys, im an IT student, and my project currently involves weighing scale. So i bought arduino mega 2560, hx711, and 4pcs of 50kg load scale (total of 200kg). now, i still dont understand how to make this thing work (to get the kilogram).. can someone please explain it to me or atleast provide a sample code?

vicatcu, 2017-06-11:
@vainell https://learn.sparkfun.com/tutorials/load-cell-amplifier-hx711-breakout-hookup-guide does a far more thorough job of it than anyone could do justice to in a github issue response.

vainell, 2017-06-11:
i tried the sample code but im still in puzzle because i dont know how to get the kilogram from it, and also on how to get the proper calibration value to pass on set_scale(); for my own load scale. i tried many calibration value but to no luck im still not getting the exact result. im sorry, im just a newbie and am also not an ECE student so i dont understand much of their terms

vicatcu, 2017-06-11:
@vainell you can use this line of code to get a raw Analog-to-Digital conversion from the HX711:

davidjonz2005, 2019-04-13:
just write down what you read from Hx711 and what is the exact load in a row for each sample .
the more sample the better precision

use Excel and make a scatter drawing by those pair point, exact measure as y and module reading as x axis.
right click on the graph and choose add trend lines ,select linear , and in the bottom select display equation in chart,

equation is like y=ax+b which means that you must multiple each reading to a and then add it by b