bogde/HX711 #21: scale.get_max

cgdavis007, 2015-12-24:
Very handy library.  It works well with lcd.print and I've tried modifying my sketch to achieve my desired goal, but I'm not a code guy so what I'm asking may not be possible.   Would it be difficult to add this to your library?  It would be nice to apply a force to the load cell and watch the numbers climb, stopping at the highest force applied.  This would be useful for a load force tester.

Many Thanks,
Chuck

bogde, 2015-12-25:
if i understand your requirements correctly, you just need to store and print the max value in your code. something like this:

```c
...
value = 0;
temp = scale.get_value();
if (temp > value)
{
    value = temp;
    ...
    lcd.print(value);
}
...
```

cgdavis007, 2015-12-26:
Your understanding is correct.  I've tried if and for statements and even while loops.  I integrated your suggestion into my code but the value returned is not in units and varies from reset to reset with no change once weight/force is applied to the load cell.  Do you have any other suggestions?  Thanks in advance.

bogde, 2015-12-28:
you can use get_units instead of get_value. the max value you're storing in ```value``` won't survive a reset unless you save it somewhere in the eeprom (see https://www.arduino.cc/en/Reference/EEPROM).

closed 2015-12-31 by bogde