bogde/HX711 #22: it is work,but can not reboot

EagleSmith, 2015-12-31:
Your code works well, but there is a problem with all HX711 connected to espduino(like nodemcu), esprduino can not reboot and boot, do you encounter this problem?
Thanks

bogde, 2015-12-31:
no, because i'm not using espduino. the code has been tested only on (a limited number of) atmega based boards. good luck with your project!

joelseverin, 2015-12-31:
HX711.cpp uses a busy-wait loop to wait for values (delivered 10 or 80 times per second), similar to a long delay(). This is not popular on the esp. I think you could get away with putting a yield() (or whatever it is called) in that loop, so that the esp can perform its other duties.

CFTechno, 2015-12-31:
you can use a  " if (scale.is_ready())" statement to read the scale . That way it doesn't go into the while loop when requesting a value

EagleShi, 2016-01-01:
hi all,thank you for your help,it is working now.

closed 2016-01-01 by EagleSmith