bogde/HX711 #23: 2's-complement error?

joelseverin, 2015-12-31:
HX711.cpp, read(), line 67:
} else if ((0x7F == data[2]) && (0xFF == data[1]) && (0xFF == data[0])) {
filler = 0xFF;
}

Let's assume we read a value of 0x800000 (min value per datasheet, i.e. -2^23 = -8388608). The idea is to invert this as HX711 presents inverted values (I assume this is the reason the inversion is performed?). It thus becomes 0x7FFFFF, and the filler is set to 0xFF per the above code snippet. It is then joined to form 0xFF7FFFFF and +1 is added, we get 0xFF800000 = -(2^32-2^23) = -4286578688. Oops, not the 8388608 we wanted.

Just skipping the above else-if would yield the correct value:
0x800000 -NOT-> 0x7FFFFF -extend-> 0x007FFFFF -plus1-> 0x00800000

I think the confusion was that +8388608 cannot be represented in a 24-bit 2's-complement format, but as it is extended to 32 bits, it should not present any problem :)

DELHIGANESH, 2016-01-11:
data[2] = shiftIn(DOUT, PD_SCK, MSBFIRST);
    data[1] = shiftIn(DOUT, PD_SCK, MSBFIRST);
    data[0] = shiftIn(DOUT, PD_SCK, MSBFIRST);
	for (unsigned int i = 0; i < GAIN; i++)
	{
		digitalWrite(PD_SCK, HIGH);
		digitalWrite(PD_SCK, LOW);
	}
	data[2] ^= 0x80;
	return (((uint32_t) data[2] << 16) | ((uint32_t) data[1] << 8) | (uint32_t) data[0]);

FabianInostroza, 2016-09-02:
Where in the datasheet is the statement that the bits on the data line are inverted?, I think that the bit inversion done in line 60-62 of HX711.cpp and the case in line 67 are wrong.

@DELHIGANESH what is the reason for the line data[2] ^= 0x80 ?

zfields, 2016-09-29:
@joelseverin You are right. I have a pull request out to fix it. Would you do me a solid and give it a once over? Thanks!

closed 2016-10-02 by bogde

joelseverin, 2016-10-02:
I was a long time ago I fiddled with this but I think it should work fine =)