bogde/HX711 #25: Precision Question

bobr3940, 2016-02-10:
For my project I am using your library along with SparkFun's Load cell amplifier.
I pulled a load cell from a small pocket scale that has a max load capacity of 100 Grams and could give readings out to three decimal places. One of the objects I am weighing in my testing is 6.385 Grams.

Using the SparkFun sample software I was able to calibrate and take readings from the load cell. The only problem is that the readings are only out to one decimal place. The sample object I am weighing is showing as 6.3 grams.

Is there a way for me to enable your library to provide a more precise reading?

bogde, 2016-02-18:
how did you calibrate? did you use your 6.385g known weight? what value are you currently passing to set_scale?

bobr3940, 2016-02-18:
I had set this aside until yesterday. When I found that the limitation was not in your library but was in the SparkFun code sample. (They truncated the data to 1 decimal place.)  As soon as I adjusted their code I found that I was able to get readings out to four decimal places, though the readings were only stable out to 3 decimal places. Thank you for creating this library.

closed 2016-02-18 by bobr3940