bogde/HX711 #26: Calibration goes for a toss on Power reset

mohanlate, 2016-02-19:
I followed the method to calibrate the scale and call tare method in setup and everything works fine. But on power reset, irrespective of weight on the scale, output shows 0, since the setup method runs scale and tare methods again.

Please suggest a workaround. What does the method Tare really do? Is it possible to store its output in EEPROM, etc. ?

Thanks

bogde, 2016-02-19:
once you find the calibration value, remove the calibration routine from your sketch.

closed 2016-02-21 by bogde

mohanlate, 2016-02-25:
[hx711code.zip](https://github.com/bogde/HX711/files/146281/hx711code.zip)

Here is my code. It is basically just as you suggested. Everything works fine and I am very happy and grateful for this great library. However, on power reset, the output reading for the weight goes back to 0Kgs. I think this is because scale and tare methods get executed in the setup call on power reset. 

Please suggest.

Thanks

bogde, 2016-02-25:
remove ```scale.tare()``` from the setup section.