bogde/HX711 #28: HX711 RETURN CODE CONTINUED

ben6711, 2016-02-21:
Don't want to let go.  Your library is terrific.  Again, we will pay a professional rate for your work.  Basically, we just want the sketch to continue even if a scale fails.  Currently,  the sketch freezes.

If you don't want to continue then,  please point us in the right direction for either making modifications ourselves (probably not in our capability) or hiring programming and  Is our request feasible?  

Ben Becker

CFTechno, 2016-02-21:
The sketch doesn't "freeze" ! Just read line 11 in hx711.cpp for understanding what is going on. "while (!is_ready());"  If you would do that test yourself before trying to read the sensor value your sketch will never stay in that while loop.

bogde, 2016-02-21:
@ben6711 i feel your use case is a very particular one and making changes to the library to make it work for your project will not bring any advantages to the other users. also, i'm a programmer, with limited electronics skills; when i code i assume the hardware part usually works as designed. what @CFTechno suggested is probably the key to your problem. you can try to remove line 41 ```while (!is_ready());``` in hx711.cpp. you may do the same check in your sketch, possibly adding some kind of timeout.

bogde, 2016-02-29:
@ben6711 is the code working for you now? do you still need any help?

ben6711, 2016-02-29:
Definitely need help.  Unsuccessfully pursued  @CFTechno's suggestion.

Need some kind of return code to indicate a  read failure.  We are running dozens of scales so if one scale fails, or is disconnected,  we either stop running or get erroneous reads which look legitimate.

CFTechno, 2016-02-29:
If you post your code that you think stops when a scale is disconnected then we can have a look and help you.

my code is `		if (scale.is_ready())
		{
			// read scale
			Lvalue = scale.read() - Offset;
}
`

If you have erroneous data you need to include code to filter out the outliers or check the noise levels on your connections.

electrokean, 2016-02-29:
@ben6711 if the suggestion by @CFTechno doesn't work, then you may need to add a pull-up resistor on the DOUT signal at the Arduino side.
Thus, if the scale is unplugged, then the HX711 pin might be floating and give unpredictable results to the is_ready() call.  The pull-up will make it always look not ready.
So code something like:
`bool failed = true;`
`if (scale.is_ready()) {  Lvalue = scale.read() - Offset; failed = false; }`
BTW, I'm presuming that when you unplug the scale, that includes disconnecting the HX711 from the Arduino.

ben6711, 2016-03-01:
if (scale.is_ready()) {  }    may be the solution.

This code seems to work with various quirks when scales are disconnected and re-connected to the Arduino.
Multiple scales are read in a loop so,  arguements in HX711 scale(... ) resolve to pin numbers. 

HX711 scale(ARG_DAT[port],ARG_CLK[port]);                                                     
if (scale.is_ready()) {
  reading = 12345;                                        //  READING FAILED:  FORCE  DUMMY VALUE.
  }
else {
  reading = scale.read_average(1);               // READING SUCCESSFULL.
}   
  
Note:   The clauses work backwards.  The else clause should handle READ FAILURES.

My  C,  C++  programming skills are limited and without your help this project would have a high exposure to failure.
Thank you!     
 I expect more help will be needed   as we progress to full scale implementation.

CFTechno, 2016-03-02:
Your assumption that scale.is_ready can be used to check failed reading is not correct. The scale.is_ready looks at the status of the HX711. That status is telling the sketch that the HX711 is finished sampling the scale electronic signal. that can happen either 80 or 10 times per second. If the scale.is_ready return false you only know that sampling is not finished, nothing more. So you can NOT use it for failed reading checking.

Using the if (scale.is_ready) routine allows the sketch to continue running it's other code because now it will not wait for the HX711 to finished sampling.

If you have (for example) 3 scales connected then imho you should code it like

HX711 scale1(port1,port1);
HX711 scale2(port2,port2);
HX711 scale3(port3,port3);
Boolean Scale1Read,Scale1Read,Scale1Read;

Scale1Read=false;
Scale2Read=false;
Scale3Read=false;
if (scale1.is_ready) {scale1_reading = scale1.read_average(1); Scale1Read=true;}
if (scale2.is_ready) {scale2_reading = scale2.read_average(1); Scale2Read=true;}
if (scale3.is_ready) {scale3_reading = scale3.read_average(1); Scale3Read=true;}

Yes, it's possible to put the ports in an array and also the scalex_reading etc but optimizing code should happen after the contour of the sketch is solid and works.

I used the Boolean ScalexRead so that any other routine now knows if it should use the value in scalex_reading

Note: read_average(1) is the same as read() as with only one 1 value there is no average.

electrokean, 2016-03-02:
@CFTechno yes, you're technically correct that the is_ready() could just indicate the HX711 is still busy, and that could happen if you're trying to read the weight more frequently than the conversion rate.  For @ben6711 he might need to use a counter or timer to determine how long each HX711 has been not ready to determine if it is malfunctioning, unless he specifically only reads the weights at a larger interval e.g 1 second.
Using a shared SCK pin isn't a good idea in my opinion as it will clock the data out of all the HX711 chips and begin new conversions forcing DOUT high till the conversion completes.  So if you check them quickly in a sequence, I think only the first will ever be "ready".
I actually stopped using the HX711 because at higher read rates (40Hz) it would seem to randomly give bad data.  I have a suspicion this is actually due to the internal buffer getting updated with a new conversion while you're still clocking out the previous conversion data.  I also needed 100Hz conversion, so I moved to ADS1232 with an external clock.
Regarding the code used by @ben6711 it also would likely be pretty inefficient and troublesome as it is apparently continually instantiating and destroying the HX711 objects.  I don't know how he can do that and keep the tare and gain correct.  The zero point on load cells will constantly drift and need some kind of auto tare on a regular basis.

CFTechno, 2016-03-02:
I agree.
```
long LastReadScale1;

if (scale1.is_ready()) {scale1_reading = scale1.read_average(1); Scale1Read=true;LastReadScale1=mills()}
```
could help in determine if a scale has issues. Nice idea!

I have no issues with the HX711 @80sps but I really have to pay attention to any possible interference. 

re-read the source and I also agree about the single clock port, will clock out the outer scales as well, not a good idea!