bogde/HX711 #29: Include begin procedure and change object constructor code

CFTechno, 2016-02-21:
From what I understand the moment a global HX711 object is created can not be controlled by the programmer. As a consequence that object might be created before the control of the pins is available. I understand that that is the reason for a lot a libraries to include a "begin" routine like Serial,begin

Because the begin will be execute after the pin control is available that is the correct moment to set any pins is need. I therefore propose to change
`HX711::HX711(byte dout, byte pd_sck, byte gain) {
	PD_SCK 	= pd_sck;
	DOUT 	= dout;

	pinMode(PD_SCK, OUTPUT);
	pinMode(DOUT, INPUT);

	set_gain(gain);
}
`

into 
`HX711::HX711(byte dout, byte pd_sck) {
	PD_SCK 	= pd_sck;
	DOUT 	= dout;
}
`

and include a new procedure begin like:
`HX711::begin(byte gain) {
	pinMode(PD_SCK, OUTPUT);
	pinMode(DOUT, INPUT);

	set_gain(gain);
}

Problem with this proposal is that it will break old code! so maybe not replace the new constructor routine such that those calling the constructor without the gain parameter will get the new constructor

bogde, 2016-02-21:
looks like a good idea, i'll think about it and let you know.

electrokean, 2016-04-10:
@CFTechno can you provide an example of when pin control wouldn't be available at the time of global object creation?  I don't see a problem with the change other than backwards compatibility, I am just trying to see what value it adds and in what circumstances.

CFTechno, 2016-05-10:
Please see the remarks made by Paul Stoffregen in
http://forum.arduino.cc/index.php?topic=45794.0
and
http://forum.arduino.cc/index.php?topic=78114.0

Setup() is the only moment within a sketch dat you can be sure that all needed initialization is done.

SirUli, 2016-06-21:
@electrokean when you want to wrap the library like shown [here](https://github.com/SirUli/Task-Examples) for other sensors, then this is very useful :+1:

closed 2016-10-02 by bogde

amotl, 2019-02-17:
Just wanted to let you know that by [1], we let the constructor-based initialization go completely. So, it can't even be used by accident anymore.

[1] https://github.com/hiveeyes/HX711/tree/spring-cleaning