bogde/HX711 #31: read function is slow

leszek77, 2016-03-21:
Hello when i use the function get_units() in my code
then the programm slows very down.
i have figured out this is caused by
long HX711::read() {
	// wait for the chip to become ready
	while (!is_ready()); <<<<<<<<<<<<<<<<<<<<<<<<<<<< HERE

i think it need to be rewritten with attachinterupt or timer?

leszek77, 2016-03-22:
I have study this code. Now I quite understand your code. The issue is that you try to get all bits in one cyclus of the main function loop. This cause trouble and the CPU Idle a lot and this slow my other code down. 
Now i have started a new approuch one bit each cycle. 

const byte hx711_data_pin = A7;
const byte hx711_clock_pin =A6;
bool  bOut,bOut1,bOut2;
bool  bIn,newData,readingHX;
byte data1,data2,data3;
unsigned long currentTime;
unsigned long lastUpdate1,lastUpdate2;
int i,d,j;
long int dd;
void setup() {
  // put your setup code here, to run once:

  pinMode(hx711_clock_pin,OUTPUT);
  pinMode(hx711_data_pin,INPUT);
 // digitalWrite(hx711_data_pin, HIGH);
  Serial.begin(9600);
  readingHX=1;
}

void loop() {

if(readingHX){
  if(bOut2){
    bOut2=0;
    digitalWrite(hx711_clock_pin,0);  
  }else{
    bOut2=1;
    digitalWrite(hx711_clock_pin,1);  
    bIn=digitalRead(hx711_data_pin); 
    j=23-i;
    if(i<24){
      dd |= bIn<<(j);     
    }
    i++;
  }
}else{
 i=0;
}
if(i==24){
  readingHX=0;
  dd ^= 0x80;
  Serial.println(dd,BIN);
  dd=0;
  digitalWrite(hx711_clock_pin,0);    
}
}

CFTechno, 2016-03-22:
Would it not been easier to use
`
 if (scale.is_ready()) Lvalue = scale.read() ;
`

bogde, 2016-03-25:
@leszek77 do you still need any help with this?

leszek77, 2016-03-26:
yes, but i try to figure out by my self.

closed 2016-03-26 by bogde

kasperv79, 2016-12-30:
Hi.
@leszek77, did you finalise your version?

leszek77, 2016-12-31:
i didn\t continue to work on it

theharshalpatil, 2017-11-07:
@bogde I'm facing the same issue. Is there a way to increase the speed at which the data is fetched?

Srijal97, 2018-09-01:
For anyone reading this in the future, that `while (!is_ready());` in `read()` caused the code to block and would waste a lot of processor cycles so I modified it by removing the while loop, and instead repeatedly checking in the main loop if chip `is_ready()`. I created separate functions `read_direct()` [Without the blocking while loop], and `get_units_direct()` [Which calls read_direct() directly] and added this to my main loop:
```
if (scale.is_ready()){
    tension = scale.get_units_direct();
}
```

MugetsuX, 2019-01-10:
@Srijal97 Hey there, i'm facing the same problem with the _while (!is_ready());_ function. I'm not a programmer and can't figure out how to change the while loop to an if function. Could you give some more detail on how you have set up your code? I'm using the multi-library to connect 4 loadcells:
https://github.com/compugician/HX711-multi/blob/master/HX711-multi.cpp

Thanks a lot :-)

Srijal97, 2019-01-10:
@MugetsuX Hey, if you could just drop me an email on the ID from my GitHub profile, I'll send you the modified library and example code showing what I mean!

jeroentjeee, 2019-01-14:
@Srijal97 Hi Sir, I tried your way of fixing this problem, however i'm not quite getting it done. could you also send me the modified libraby?

thanks a lot

andrelouk, 2019-02-24:
@Srijal97 could you send me your modifications to the library. thank you very much.. im doing a rapid acquisition from the load cell, and i need to get the value as fast as possible. Thank you very much..

Srijal97, 2019-02-24:
> @Srijal97 could you send me your modifications to the library. thank you very much.. im doing a rapid acquisition from the load cell, and i need to get the value as fast as possible. Thank you very much..

Please find the files here: [Google Drive - HX711](https://goo.gl/Ys3Ycv)

andrelouk, 2019-02-25:
Thank you Very Much... Really Appreciate it

eleksis, 2019-03-15:
@Srijal97 thanks for sharing the code. If i can suggest...you should make fork of this project and publish your code on your github profile, that's what github is for...

Srijal97, 2019-03-17:
> @Srijal97 thanks for sharing the code. If i can suggest...you should make fork of this project and publish your code on your github profile, that's what github is for...

I have done that now, thank you!