bogde/HX711 #32: Major issue

mohanlate, 2016-03-24:
Wither this is a major issue, or I am missing something here. Here is a simple code -

```
#include "HX711.h"
// HX711.DOUT	- pin #A0
// HX711.PD_SCK	- pin #A1
HX711 scale(A0, A1);		// parameter "gain" is ommited; the default value 128 is used by the library

void setup() {
  Serial.begin(9600);
    scale.set_scale(119000);
   scale.tare();
}

void loop() {
  Serial.println(scale.get_units(10));
}
```

The scale parameter is worked out using the method suggested by you and it works perfect!

The problem is when there is a power/arduino reset and weight is still present on the scale(load cell). The output goes back to 0 and I have to do the following again:
1. Remove the weight from the load cell
2. Power reset
3. Place the weight back on the scale.

Then it shows correct reading again.

Problem is that my weight is 200Kgs and it is not practical to remove weight and reset system each time there is a power failure / reset.

Please suggest code changes. Since this is a commercial application, I am ready to pay for your solution.

Thanks in advance. The library is great!

bogde, 2016-03-25:
Did you try to remove `scale.tare();` from your `setup()` section?

mohanlate, 2016-03-25:
When I do, I get wrong results. I get 77 even when there is not weight. And when I put weight, I still get 77.

electrokean, 2016-03-25:
Just like you've calculated your scale value, you need to calculate your offset (e.g. 77).
You then manually set it in `setup()` section using `set_offset(77);`

bogde, 2016-03-25:
that's correct. also, you can add something like a tare button to your device, read here to see what i mean by that: http://www.escali.com/blog/using-the-tare-feature-on-digital-kitchen-scales/

you can also add code to save the last weight value, offset etc., so in case of power loss or reset you can start with the last known values. that plus a tare button and maybe a way to set the offset using some kind of input would give you more flexibility.

electrokean, 2016-03-25:
Indeed - in my systems I store the offset in EEPROM, and can send a serial command to tare at any time.  I also have an auto-tare function in my host code that triggers when the weight (and noise level) is below a certain threshold.
This compensates for drift over time and temperature, which is normal in load cells, but also accumulation of debris (sand & penguin droppings in my case).  You'll need to determine if your accuracy requirements need that complexity.
Some good info on drift & noise here: https://phidgets.wordpress.com/2014/05/12/noise-and-drift-in-load-cells/

bogde, 2016-04-06:
@mohanlate do you still need any help with this?

mohanlate, 2016-04-06:
Thanks for all the valuable help. I went back to using the original library
for the time being since it served my basic purpose.

closed 2016-04-06 by bogde

JoeToeniskoetter, 2019-05-08:
> Just like you've calculated your scale value, you need to calculate your offset (e.g. 77).
> You then manually set it in `setup()` section using `scale.set_offset(77);`

Could you help me figure this out? How do I go about calculating the correct offset?