bogde/HX711 #33: Arduino, hx711,

dmurphydrtc, 2016-03-28:
Thanks for sharing your time and software.

I am using a 100/150kg load cell and a green hx711 ADC purchased from Aliexpress.
I have made my own scales using the above load cell (want to measure bee hive weight) and have hooked all the various components and attached my Arduino (Freetronics Eleven) to my Mac Air.

I am a new noob but have played with the Arduino and understand how to use and set it up.

My problem is that when I run the hx711 code my readings are 2 decimals places out. ie If I place 8kg weight on the scale (after waiting a minute to see a reading of 0g) and then place my weight on the scale I get 82.0g (expected 8200g) or if I place a 2kg weight I get 21.1g (expected something like 2000g).

I am running the code as is, using channel A on the hx711 but at a loss as to what I am doing wrong?
Looks like an issue with voltage or reference resistance. Tried 2 different hx711 units.

Can some shed some light? Thanks again.

bogde, 2016-03-28:
did you follow the calibration procedure? what weight did you use?

electrokean, 2016-03-28:
As bogde mentions, this sounds like a calibration issue.
You effectively need to reduce your value passed to set_scale() by a factor of 100.
Are you using get_units() which returns the scaled weight, or get_value() which returns the raw value?

dmurphydrtc, 2016-03-28:
Kean/Bodge,

I obviously stuffed that up, will work through my calibration again and
report back.

Thanks.

dmurphydrtc, 2016-04-06:
Bodge,

Just have not had time to get back to it...Hopefully in the next night or
2. Thanks again.

bneedhamia, 2016-04-14:
If you aren't aware of it, check out Nate Seidle's beehive scale project, at http://makezine.com/projects/bees-sensors-monitor-hive-health/ although he doesn't go into detail about calibration, he describes a lot of other details.

I just recently blogged about how to calibrate an "old school math" way, for my Dog Bed Weight Scale project.  See https://needhamia.com/?p=720 for a step-by-step of how I calibrated it.  The raw numbers I read and formulas I used are in a Libre Office spreadsheet is at the github repo of that project, at https://github.com/bneedhamia/CurieBLEWeightMonitor

electrokean, 2016-04-14:
@bneedhamia nice writeup there, and good to see you doing some statistical analysis.  I actually have calculations built into my sketch to calculate noise and ENOB (effective number of bits) on command.
I think you biggest issue you have (per your blog) is that the "load cells" you're using just aren't particularly good, thus leading to mismatches between cells, which totall screws things up when you wire them together.  You don't need expensive load cells, but it isn't difficult or costly to find some better than those.
Get some proper 4-wire (or 6-wire) beam style ones and mount them solidly, wire them in parallel (although test them individually first) and you should get significantly better results.
FYI, 6-wire ones just separates the excitation and excitation sense, to remove voltage drop effect from long or thin wiring.  The HX711 doesn't support 6 wire load cells, but I guess you could use channel B to measure any error, and I suspect that is why the channel B input exists.  Otherwise you just connect E+ and Sen+, and E- and Sen-.  Note that Sen+/Sen- is different from S+/S- (signal).
Hope this is of help.

closed 2016-10-02 by bogde