bogde/HX711 #34: Readout Consistently reading 0

diattamo, 2016-04-01:
Hi, 

We're working on a project trying to get the readout using 4 load cell sensors. So we're running the code you provided in the examples folder. However we keep getting a 0V voltage readout. We think it might be due to the readAverage value, but we don't even know how that is computed.  

Here is the output we get:
[image]

Any help will be very much appreciated.

Thank you.

electrokean, 2016-04-01:
Please provide more info - is this just the demo code?  Have you modified anything?
But I think you have a wiring issue - can you provide wiring diagram and/or photo?
If you have 4 load cells wired in parallel but incorrectly it could possibly cause this.

diattamo, 2016-04-06:
Our problem was due to a mistake in our wiring. Thank you!

closed 2016-04-06 by diattamo