bogde/HX711 #35: Seting Up multiple amplifiers for multiple load cells

rtlop, 2016-04-07:
So the library works great for one amplifier and load cell. But what would I need to change if I wanted to set up to 2 load cell and amplifiers?

electrokean, 2016-04-10:
Two ways I can suggest:
1) use channel B to connect the second load cell and select it using set_gain(32)
2) instantiate a second HX711 object using a different pair of I/O pins, and use a second HX711 interface PCB, both using channel A for the load cell input

Note that option 1 has some limitations (limited gain and limited library support), and relies on your HX711 interface PCB breaking out the channel B input pins.  You would use the same excitation output for both load cells.

bneedhamia, 2016-04-14:
I'm building a scale using 4 load sensors (each is half of a load cell) and Sparkfun's HX711 board (which has only one set of inputs).  I recently decided to use a separate amplifier and HX711 object for each load sensor so that I can separately calibrate each load sensor.  (Of course I'm adding two resistors to each load sensor to turn each sensor into a Wheatstone Bridge.) It's good to hear someone recommend using multiple HX711 objects in one Sketch.

electrokean, 2016-04-14:
@bneedhamia you mention adding two resistors to create a wheatstone bridge from those half-load cells.  What tolerance resistors are you using?
Your blog post mentions 0.7lb variance for approx 15lb weight in different locations.  That is just under a 5% variance.  I hope you're not using 5% resistors :)
Even if you use a pair of 1% resistors, you could see +/-2% variance or approx 4% variance between any two of the bridges (not even allowing for the load cell matching).

pit001, 2016-04-25:
Hello, 
I use for test three HX11 with load cells. 
And this code:
HX711 scale0(A1, A0);	
HX711 scale1(A2, A0);
HX711 scale2(A3, A0);
#define SENSORCNT 3
...
void setup()
{
HX711 *scale[SENSORCNT];
...
}

void loop()
{
for (uint8_t i = 0; i < SENSORCNT; i++)
  {
long read = scale[i]->read();
value[i] = (read - scale[i]->get_offset()) / scale[i]->get_scale();
Serial.print("  ");
 Serial.print(read);
Serial.print("  ");
Serial.print(value[i], 1);
}
...
}

matouchat790te, 2016-09-05:
I have the same issue as pit001, with one XH711 it's ok. With 3 , sometime values not valid.  An idea ?

electrokean, 2016-09-05:
@matouchat790te Are you sharing a pin for the HX711 clock like A0 in the above example?
I believe this could be the cause.  Can you try wiring two separate pins for each HX711.
You don't want to be clocking all 3 HX711 chips every time you read one, as it could lead to this kind of issue.

matouchat790te, 2016-09-05:
I'm not sharing pins. each HX711 has is own . 
If I do a squale_power_up / down at each loop, the problem disappear. But it's too slow.  
One other way is to use three arduino board and 'OR' the ouputs, but it's not elegant . the project is for a touch probe for a cnc milling machine.

matouchat790te, 2016-09-05:
I'll put an oscilloscope on the pins, and a logic analyzer to see if the problem is outside or not.  may be I have crosstalk between channels which makes pulses on the wrong clock input.  Disable interrupts has no effect.
I'm glad to see I'm not alone with this issue.

matouchat790te, 2016-09-12:
Hello all. 
I did some measurements 
And I noticed an important point: the HX711 periodically performs conversion without external application need. It works in free running.

With only one HX711 : If we wait for the conversion end signal, all goes well. ( Falling edge of DOUT )  This works for one XH711.  
But as soon we use more, calling the ‘is_ready’  function becomes asynchronous. It is not clear in the datasheet, but the HX711 is free running at 80Hz. And if you activate CLOCK signal a few microseconds before the next end of conversion, you may just send the clock at the wrong time in the 10 microseconds time . Named  # T1 1: falling edge to DOUT PD_SCK rising edge. (See datasheet). and data will be wrong.
In all the tests I made the datas were false if the Arduino make a request exactly 12.75ms after the last end of conversion signal. ( falling edge of DOUT ) 
This bug could also appear with one HX711 if you make asynchronous read of the chip. 

I have an imperfect workaround: make a loop for the FALLING edge of DOUT rather than checking its level and immediately start the clock for reading data.  But you may be waiting 12 ms more than necessary.  

Anyone have a better solution ?

electrokean, 2016-09-12:
@matouchat790te Yes, I experienced this behaviour too.  My original debug code was doing too much processing or serial output between readings.  Even after some work I would occasionally still get an incorrect reading which I would have to filter out.  In the end I gave up on the HX711 due to its limitations and poor documentation.

bneedhamia, 2016-09-12:
I've had nothing but good luck using the HX711 board from Sparkfun and 
the https://github.com/bogde/HX711 library, connected to an Arduino 
101.  I used them in both a dog bed weight scale and a dog water bowl 
scale.  My app code is at 
https://github.com/bneedhamia/CurieBLEBowlScale and 
https://github.com/bneedhamia/CurieBLEWeightMonitor if you'd like to 
take a look.  The Bowl project uses one load cell connected to the 
HX711; the bed uses 4 load sensors (each 1/2 a wheatstone bridge) plus 
matched resistors, and 4 HX711s.

Granted, all those projects do is read the weight and send it via BLE to 
a Raspberry Pi gateway - they don't do a lot of computation between 
HX711 readings.

matouchat790te, 2016-09-12:
One way could be using interrupts , one on each HX711. so moving on Mega . 
I found a similar AD with same serial link here : http://www.analog.com/media/en/technical-documentation/data-sheets/AD7780.pdf
they spoke about 'ensure reading after new conversion signal '

compugician, 2016-09-16:
Hi all,

I wrote a library that resolves most of the issues discussed here with multiple cells, while also sampling all of them at 80hz simultaneously.

check it out:
https://github.com/compugician/hx711-multi

It is stripped at the moment of all the other functionality (tare, offset, units, etc.) but those are easy to implement on your own in or out of the library if you'd like.

I'll add them soon, hopefully, or if someone wants to contribute please do.

matouchat790te, 2016-09-18:
Hello all, 
I replaced in the beginning of function read() 
long HX711::read() {
	// wait for the chip to become ready
	while (!is_ready());
-------
by this 
long HX711::read() {
	// wait for the chip to become ready
	if (!is_ready()){ 
	   while (!is_ready()) {  // on est à 1 , on attend front descendant
	  	// Will do nothing on Arduino but prevent resets of ESP8266 (Watchdog Issue)
	 	yield();
	   }
	} else {
		   while (is_ready()) {
	  	// attend 1 
		 	yield();
	       }
		   while (!is_ready()) {
	  	// attend front descendant 
	 	yield();
	   }
	}
___________
waiting the FALLING edge of data ready ( DOUT )  ensuring the data is available for the next 10 ms

compugician, 2016-09-19:
@matouchat790te 

I'm looking to understand this problem better so I can fix it also in my copy of this repository. 

If I understand your code correctly, doesn't that effectively reduce the sampling rate by *at least* half?

Maybe I'm misunderstanding the original issue, but if the issue that some samples are bad because we're sampling too 'late' after a ready signal, maybe the solution should be to check the timing from the last known good sample, and as long as it is within a threshold of a full cycle we go ahead and sample, and otherwise wait?

Of course that is not guaranteed to work forever because of the two-clocks problem, but that could be mitigated either by once in a while re-setting the chip, or otherwise re-synchronizing our timing (to a falling edge, for example, by using your trick)?

Thoughts?

SreevatsaAcharya, 2017-07-19:
Hello,
I am trying to send the value obtained from the load cell to a server using a gprs sim900 module, and i am doing so as follows
```
char val[20];
char url[160];
float temp;
temp= scale.get_units(); // or scale.get_units(10); AVERAGE OF 10

  dtostrf(temp, 7, 2, val);
  snprintf(url, 74, "AT+HTTPPARA=\"URL\",\"http://www.website.com/test.php?temp=%s\"", val );
  mySerial.println(url);

```
and sending the variable "temp" in the http url request. i get the value on the serial monitor when i print the variable on the serial monitor but it does not get updated on the server. I just get a blank reading.

Am i missing something? Any solutions?

compugician, 2017-07-19:
I'm not sure how this is a question to the HX711 issues forum.

1) The most obvious thing not shown in your code is how you are sending this to the http server.
2) Do you know your server works? If you manually send a value in a URL of an http request does it work?
3) If yes to #2, use the GPRS_TCPConnection example (or equivalent that came with the library for your shield), and manually change the URL there to yours. Notice that if you are using the seeed-studio library then you split the connection into gprs.connect to the server and gprs.send with a get command such as "GET /test.php?temp=123.45" (or put, or whatever your PHP script is expecting).
4) If yes to #3, and indeed you see the value in 'val' when printed out, then check how your code for sending the http request differs from #3.

SreevatsaAcharya, 2017-07-19:
@compugician 

`
float temp;
char val[20];
char url[160];

void SendData() {
  mySerial.println("AT+HTTPINIT"); //init the HTTP request
  delay(2000);
  ShowSerialData();

  //  Serial.print("Reading: ");

//  temp=random(1000.12,9999.99);
  temp= scale.get_units();
  Serial.print(scale.get_units(), 1); //scale.get_units() returns a float
  Serial.print(" Kg"); //You can change this to kg but you'll need to refactor the calibration_factor
  Serial.println(temp);
  
  delay(3000);

  dtostrf(temp, 7, 2, val);
  snprintf(url, 74, "AT+HTTPPARA=\"URL\",\"http://www.website.com/test.php?temp=%s\"", val );
  mySerial.println(url);

  delay(3000);
  ShowSerialData();

  mySerial.println("AT+HTTPACTION=0");//submit the request
  delay(10000);
  ShowSerialData();

  mySerial.println("AT+CIPCLOSE");//close the connection
  delay(5000);
  ShowSerialData();
}`

This is the code i am following and i am able to send static data to the server i.e if i define temp=2134 etc in the url. It also works if the variable temp receives the value from a random number generating function as shown. 
But when i transfer value from the scale.get.units() function it returns blank.
I am using a Sim900 rs232 gprs module and found this method to be most suitable to send data. Initial settings are all correct and verified.

I tried the GET method but faced a lot of issues so shifted to this method.

compugician, 2017-07-19:
So if you set temp manually it works, but if you use scale.get_units() it does not?

What is printed by 'Serial.println(temp)' in each case?

Also, what random() funtion are you using? AFAIK the standard one does not generate floating point numbers.

Does rounding temp solve result in the data getting to the server appropriately?

SreevatsaAcharya, 2017-07-20:
Finally!!! Made it work. The scale.get.units() function reads value from the load cell very quickly so i guess the value wasn't actually getting stored in the variable i defined. Made a few changes , and now it works. 
Thanks for your help @compugician

lucian0112, 2017-09-27:
@compugician - First of all great job with the library.
Is there any chance to include in it the tare function and calibration? 
I tested your library with 12 hx711 sharing same clock and seems work fine, but I was not able to implement these functions which are actually very important for me....

Thanks

jpk73, 2017-09-27:
check out these examples:
https://github.com/sparkfun/HX711-Load-Cell-Amplifier/tree/master/firmware

tk5ep, 2017-10-18:
Hi,
@lucian0112 
Do i understand it correctly that you're using a  DOUT pin common to all your 12 HX711 ?

I would like to use 4 HX711 using the minimum number of pins of my pro mini. I don't need any speed, so i will use 40Hz not 80Hz. 

Would the original library work without here described problems ?

Thanks,

lucian0112, 2017-10-18:
Hi,
Yes you are correct. I used that lib and 12 hx711 with clock pin shared. The problem for me was that I was not able to implement functions in it like (tare, calibration etc...). The lib just read raw values of the chips.
Not sure if is relevant but my hx's were modified for 80Hz.
Meanwhile I moved on arduino mega and have enough pins to use them on classic wired 

Cheers

zstergios, 2017-10-26:
Hi there!
I have work in past with1 load cell and + HX711 (bodge library)

I want to build a large scale, so I will need 6 load cells. (2 front,2 middle,2 back)
I planning to use 3xHX711 chips (1 per 2 cells/row)

My purpose is to get the total weight.
I have checked the multi-HX711 library, but what's the logic of combining 3 values?
I guess it's SUM of readings divided by 3 (total of amplifiers)?
How accurate is this way?

THANK YOU

tk5ep, 2017-10-29:
Hi,

How do you connect 2 cells to one HX711 ?

Never thought about your problem, but i don't see any other way to solve it than summing the 3 readings.
About accuracy, if you have a common scale, you will never know how much weight is on which cell. You can only suppose it is an equal part of the total weight... so ....

Keep us informed, i might have the same need soon !

zstergios, 2017-10-29:
You can connect 4 load cells to one chip.
Search for half bridge and full bridge hx711.

The issue is when you have 6 or 3. Need to summarize them probably

lucian0112, 2017-11-03:
Have you connected 4 load cells to one chip? Please tell us how you did it.

zstergios, 2017-11-06:
It's simple, just search on google.
https://i.stack.imgur.com/9vDyI.jpg (Source http://prntscr.com/h6u02q)

lucian0112, 2017-11-06:
In your link there are 3wire load cells, I was talking about 4 wire load cells like this 
https://cdn.instructables.com/F6R/MFZR/IU9UQLN9/F6RMFZRIU9UQLN9.MEDIUM.jpg

Also the idea is to read each load cell individually .

zstergios, 2017-11-06:
If you read the individually it's
hard to have good accuracy & calibration!

I think (NOT SURE), you may have to ignore on cable.

I have only one load here with 4 wires, I will order some more to test it.

Stuckcesium, 2017-11-10:
I'm also interested to plug many HX711 (up to 8) to an arduino pro mini (only 4 analog pin). 
What do you think with a solution using a multiplexer like a 4051 and the library developed by  @compugician ?

lucian0112, 2017-11-10:
You cannot use a multiplexer. If you read hx711 datasheet and read the library you will understand why.
Also why are you thinking only on analog pins? It works very well on digital pins also....so for 8 hx you need 16 pins not matter analog or digital...

Stuckcesium, 2017-11-10:
Thanks Lucian for you reply, 
I didn't know the HX711 was working with digital pin... So good news for me. I've have to learn a lot again on arduino and electronic in general...

lucian0112, 2017-11-10:
you are welcome
good luck :)

zstergios, 2017-11-10:
@lucian0112 have you figured out how to connect 4-wire load cells together?

lucian0112, 2017-11-10:
I did a test and just connected them in parallel and used only one hx but this is not what I want. I need read them individually .
Or this was not your question?

zstergios, 2017-11-10:
I thought we need the same.  4 load cells (4wire each) connected together and getting 1 reading/weight instead of connecting each of them individually and getting 4 readings/weights.

If you connect them  individually:
1)   how you will calibrate all loads?
2)  how you will get a final weight?

lucian0112, 2017-11-10:
I calibrate individually them and final weight is sum of all cells if you want this.
In my case I need to measure 12 individual applied forces so that s why I need them read individually ...

zstergios, 2017-11-10:
Nice.....I will try it.
Do you think will it have accuracy?

lucian0112, 2017-11-10:
accuracy depends of how good you do the calibration. anyway here can be a long discussion.
what are you trying to achieve exactly?

zstergios, 2017-11-10:
Large Scale with 6 or 9 load cells. Size 1.3mx 1m

lucian0112, 2017-11-10:
For this size 1.3 x 1m I would use only 4 load cell, one each corner. Choose load cell that support the weight you need to measure. If you want to measure big weight you should proper make the frame but I don t see any reason for use 6 or 9 cells....   
do you have a draw of the scale?

zstergios, 2017-11-14:
Red are the basic 4, the green are the optional. 

1) http://prntscr.com/hahcur Design with 4+2
2) http://prntscr.com/hahdp4 Design with 4+5

I want to measure palette, so because the weight is around 2T, and my fear not to break.,
that's why of design (2)

The gaps are required the clark to able to lift and load the scale 
https://a.scdn.gr/images/sku_images/014957/14957290/20160520150235_1ad5a025.jpeg

My problem right know is not the amount of load cells 4 or more but **HOW i can connect them.** to able to get the **TOTAL weight**

I have spent many days but no luck. Most suggest to add HX711 as many cells I will use (individually)

I asked a manufacturer on Alibaba and told be 2 usefull things
1) The total weight is divided by the amount of cells, for 2T I need 500KGx4 Cells, but suggested me to use 1T because the weight will never be balanced on the center. I think it's obnvius
2) Told be that I need Load Cell Junction Box https://www.aliexpress.com/wholesale?catId=0&initiative_id=SB_20171114115757&SearchText=Load+Cell+Junction+Box . Something similar with HX711 but multi-cell supported. The problem is that are expensive and probably I will not able to use the arduino. I tried to searched for shematics if I able to understand how the connections made but I'm not experienced so much.

lucian0112, 2017-11-15:
Do you have skype?

zstergios, 2017-11-15:
SkypeID: zstergios
Email: zstergios [at] gmail.com
Thanks for your interest!

technicalpriyanshu, 2018-03-19:
@compugician this works fairly good with 3 load cell, but not with 4. can you suggest changes in case of 4 load cell wiring up together.

compugician, 2018-03-19:
On which platform?

There are platform-dependant optimization that need to be made, and I could
help as I've gotten it running with more than 4 on several platforms - but
didn't have the time to put them neatly into the codebase, so it would be
nice if you can, once it works :-)

rs810, 2018-04-25:
@compugician How many can you run with an Arduino Uno?

compugician, 2018-04-26:
At what sample rate?

rs810, 2018-04-27:
The sample rate can be quite low: once a second.

compugician, 2018-04-27:
You should be able to run a dozen or so with no additional hardware. And
virtually hundreds if you use additional hardware.

The number is smaller if *simultaneous* sampling.

How far apart can cell 1 sampling be from cell N sampling? (Within a given
1hz sample)

Notice that some code modifications are necessary.

rs810, 2018-04-27:
The project requires monitoring of five load cells and ideally getting an update on each one once a second. It seems we could offset sampling of each one by 200ms or groups by a larger time interval.

compugician, 2018-04-27:
> AEYFiMqUCNFdcJuJGkMKDWzG7faul8fyks5tsMtZgaJpZM4ICbpk

You should be able to do that easily even with the standard library. The
hx711 multi library (https://github.com/compugician/hx711-multi) will save
you some wires but is not really necessary if you are ok with samples being
more than 100ms apart.

rs810, 2018-04-27:
Thank you for taking the time to answer my questions.

compugician, 2018-05-17:
@lad97 - could you share a bit about what you're building? I would love to hear about what others are working on that needs this capability.

lad97, 2018-05-22:
@compugician Well to be honest, I'm not really using the capability of the multi-library you developed.. I can see a need for faster sampling in a later prototype though.  I'm working on a project for an inventory management system that uses load cells to track part counts in various bins.  Using a 1kg load cell I can get an exact part count for parts weighing only a few grams each.  For now I'm using the esp8266 (nodeMCU) board to report the part count of 3 bins to a web dashboard - simple enough demo.  But eventually the idea would be to log 30 or so different parts using a single esp8266 and the accompanying electronics/battery. And then having hundreds of the esp8266's reporting part counts to a single web interface for the inventory company.

makay68g, 2018-06-10:
@technicalpriyanshu - I had the same experience, and here is the solution.

@compugician - I found the problem, why your HX711-multi code works with 3 cells, but not with 4. According to the specification of the HX711 (page 5 of www.mouser.com/ds/2/813/hx711_english-1022875.pdf) the clock high time can take at most 50 microseconds during reading the data. I have an Arduino Nano with 16 MHz clock speed, so I have 800 time ticks to read all HX711's connected to it. It turns out that only 3 cells can be read safely in this amount of time with your code. I was able to speed up the code, so instead of

    int i;
    ...
    for (i = 0; i < 24; ++i) {
        digitalWrite(PD_SCK, HIGH);
        if (NULL!=result) {
            for (j = 0; j < COUNT; ++j) {
                bitWrite(result[j], 23-i, digitalRead(DOUT[j]));
            }
        }
        digitalWrite(PD_SCK, LOW);
    }

I used 

    long int i;
    ...
    for (j=0;j<COUNT;++j)
        result[j] = 0;
    for (i = 1 << 23; i > 0; i>>=1) {
        digitalWrite(PD_SCK, HIGH);
        for (j = 0; j < COUNT; ++j)
            if (digitalRead(DOUT[j]))
                result[j] |= i;
        digitalWrite(PD_SCK, LOW);
    }

[I know, the (NULL!=result) checking is not in this code, but if we are short of CPU time and one does not need the result, than why is he calling the read funtion anyway? :) ]

This change makes it work for 6 HX711's.

@technicalpriyanshu - If you need only 4 HX711's, then change the above code to what I wrote, and it will work. Moreover, if you never need any other number of HX711's, then unfold the for cycles, and it will run even faster. Or you can buy a faster Arduino (like Due) or even a Raspberry Pi. Or you can go for Assembly... :)

BTW: I am using load cells and HX711's in a warehouse automation logistic system. The cells will measure the weight of the commissioned goods. We will have 8 scales on each position, each scale will have 4 load cells, so I will go with two Raspberry Pi's reading 16-16 HX711's (they will have other tasks too, like showing information on a screen, so it is not an overkill :) ).

I hope this helps... :)

stadlor, 2018-09-19:
HI

I'm currently trying to run 4 load cells connected to 4 HX711 boards, with preferable full speed (~80Hz). It did not work with HX711-multi as mentioned above. I therefor tried your replacement code  @makay68g.

That is, I replaced some code in the readRaw() function, as shown in the code snippet below. However, this code dit not work for me since I'm only getting zero output. Is there anything I'm doing wrong?
```
void HX711MULTI::readRaw(long *result) {
	long int i,j;
	// wait for all the chips to become ready
	while (!is_ready());

	for (j=0;j<COUNT;++j)
		result[j] = 0;
	for (i = 1 << 23; i > 0; i>>=1) {
		digitalWrite(PD_SCK, HIGH);
		for (j = 0; j < COUNT; ++j)
			bitWrite(result[j], 23-i, digitalRead(DOUT[j]));
			if (digitalRead(DOUT[j]))
				result[j] |= i;
		digitalWrite(PD_SCK, LOW);
	...
```
Thanks for all the help in this thread!

compugician, 2018-09-19:
> I'm doing a similar project as @rs810 <https://github.com/rs810> using 3
> load cells and 3 HX711 boards. It is great to find an active forum
> discussion on the topic. Pleased with the discussion. Thanks @compugician
> <https://github.com/compugician> @lucian0112
> <https://github.com/lucian0112> @matouchat790te
> <https://github.com/matouchat790te> for you input/work specifically!

Could you share some more about your usecase?

[105 more replies]