bogde/HX711 #36: Modified read_average()

tk5ep, 2016-05-10:
Hi,

Would it make sense to modifiy the read_average() to make a moving average filtering, like this ?

long HX711::read_average(byte times) { // modified for moving average filtering
	long sum = 0;
	long min; // lowest value read
	long max; // highest value read
	long last; // last value read
	
	for (byte i = 0; i < times; i++) {
		last = read();
		if ( i == 0 || last < min ) min = last; // init and set min value
		if ( i == 0 || last > max ) max = last; // init and set max value
		
		sum += last; // sum of all readings
	}
	sum = sum - min; // reject the smalest reading
	sum = sum - max; // reject the greatest reading
	
	return sum / (times - 2); // calculate average value from leaving values
}

tk5ep, 2016-05-29:
No one interested ?

chaeplin, 2016-05-29:
I use this lib for moving avg,
https://github.com/MajenkoLibraries/Average

electrokean, 2016-05-29:
@tk5ep that code is not calculating a moving average, it is just a filter for min & max value.  The code linked by @chaeplin implements a proper moving average (& other statistical functions), but will potentially use a lot of program space, which in an Arduino or similar microcontroller environment is precious.

I would suggest that this kind of math is best left out of the library, and added by the user depending on their needs.  Some info added to the wiki regarding filtering, calculating std dev. for confidence level, etc, would be appropriate.

tk5ep, 2016-05-29:
You're right.. It's not a moving average calc. I wrote it this way, to be fast and use no program space.
I use it actually and it works just fine for me.

Just my 2 cents and it might help.
Cheers,

bogde, 2016-09-23:
i'm going to close this, but feel free to re-open as needed.

closed 2016-09-23 by bogde