bogde/HX711 #39: Has anyone tried to modifiy this library for a HX712 ?

tk5ep, 2016-06-06:
Hi all,

I would like to try to use the HX712 (i ordered some) which seems to be much better than the HX711 in different points.
I found only a datasheet in chinese, none in english. The AVIA support asking for this later, never answered to my mails... 

I suppose the library can be adapted to the few differences between the chips.

Has anyone worked on this ?

Here the link to the chinese datasheet :  [http://www.aviaic.com/image/hx712_bf.pdf]

Thanks,

bogde, 2016-06-06:
This looks like the English datasheet: https://www.scribd.com/doc/94396103/Hx712-English-Datasheet

Where did you order your parts? I'd try it out, if time permits.

tk5ep, 2016-06-06:
Hi, thanks for fast answer.

Fine, if you would like to work on the library !

As far as my knowledge permits, i will try to work on it. It seems that without any mod, it could work at 10 Hz and gain = 128 .

I ordered on Aliexpress. These chips are quite cheap 
[http://fr.aliexpress.com/item/Free-shipping-5pcs-lot-HX712-electronics-said-ASIC-SOP14-new-original/32566728886.html](url)

OK for the link, but i HATE to subscribe to any site, only for getting some infos... If you already downloaded the datasheet, can you send it to me ?

Cheers,

bogde, 2016-06-06:
Here you go: http://bit.ly/hx712

I couldn't find any breakout boards though, and I really don't have enough time to fiddle with the soldering iron and all, so I guess I'll pass for now. Good luck!

bogde, 2016-09-23:
any progress with this? just wondering.

tk5ep, 2016-09-24:
Hi Bogde,

Unfortunately not. The summer season is not the best for staying at home
and behind the keyboard or soldering iron...

I modified the library and i will soon start building a board.

See you soon,

Darktidelulz, 2017-02-02:
Tk5ep / Patrick,

How is the board comming along?

Designing a beehive scale system and would love to use the Hx712. Don't really have the skill, time nor money at the time to do it within this project.

bogde, 2017-04-08:
just wanted to let you know that i did some tests today with the HX712. the existing code pretty much works with the new chip (gain set to 128, i didn't try 256 yet). i'm going to design a board, to be able to do more tests easier.
 
[image]

Darktidelulz, 2017-04-08:
Awesome, keep us updated!

tk5ep, 2017-04-09:
Great news !

I did modify the library but did not test it... Too many things to do... :-(
Keep us informed if performances are really better than the HX711. (what
should be !)

Cheers,
patrick.

bogde, 2017-04-09:
i'm going to create a PCB based on the schematic in the data sheet. i was thinking to add an LED on the clock line, something like this:
[image]

anybody thinks that's a bad idea? i did test and it seems to work, but let me know what you guys think. of course, i could design the PCB that way but not solder the LED, i was just wondering if it makes sense to add it.

Darktidelulz, 2017-04-09:
It consumes more power, and it does not really make sense to put it on there (other than testing the design or sck line).

electrokean, 2017-04-09:
Adding any extra load to a data/clock line is generally a bad idea for signal integrity.
If you need to monitor your code during debugging, just add an LED on another free I/O pin.

bogde, 2017-04-10:
got it, thanks, no LED it is.

tk5ep, 2017-04-10:
Hi !

I agree what has been already said about the LED.
But i would recommend to add a GOOD voltage regulator for the ANALOG part.
It's the key for a stable measurement !

Cheers,

bogde, 2017-04-10:
all suggestions are welcome. anything specific you have in mind? should i have separate analog and digital voltages? the hx711 boards i have don't seem to have that. also, what about the BAT pin? that can be connected to the higher of AVDD or DVDD when not used, should i use some jumpers or just connect it to one of those?

Darktidelulz, 2017-04-10:
The hx712 measures the voltage from the battery and uses that voltage for the loadcel. Don't know if the other parts of the curcuit need a voltage regulator.

tk5ep, 2017-04-10:
Hi Bogde,

The HX712 has no internal voltage regulator like the HX711. An external
voltage regulator must be used.
I use a LT1761 which is a low current, low noise regulator. It comes in
various output voltages and those interesting us are 3.3V and 5 V.

I use the HX712 internal switch output VOUT to shutdown this regulator with
the LT1761 SHDN pin.

The digital power DVDD uses the same voltage as the Arduino. This way, you
have a complete separation between analog and digital voltages keeping
noise as low as possible.

The pcb must have a ground plane under the chip like in the board example
in the datasheet.

I couldn't understand how the BAT pin is used... The doc is very unclear
regarding this point... If somebody has more infos.. ?

Cheers,
Patrick.

Darktidelulz, 2017-04-11:
http://www.theorycircuit.com/how-to-connect-load-cell-to-arduino/

In the comments * HX712 is a precision 24-bit analog- to-digital converter (ADC) with on chip load-cell power supply switch and battery voltage detection.

**From datasheet**  
Features
*Selectable differential input or battery detection input

**Pin description**
Battery Voltage Detection Input

**Analog Input and Battery Detection**
Differential inputs are designed to interface
directly with a bridge sensor’s differential outputs.
It can be programmed with a gain of 128 or 256.
When 5V reference is used, these gains
correspond to a full-scale differential input
voltage of ±20mV or ±10mV respectively.
Single-ended battery input signal (BAT) can
be directly connected to a battery output
(2.6~5.5V). On-chip 56kΩ and 3.2kΩ resistors
divide this input voltage to provide suitable
voltage to the ADC, which has a gain of 1. When
detecting battery input, AVDD is used as the
ADC’s data conversion reference voltage. When
battery detection circuitry is not in used, a MOS
switch is used to turn off the resistor divider path
current.

**Power Supply Options**
DVDD can be connected directly to AVDD
power supply. But it should be the same or similar
voltage as the MCU power supply to ensure
proper communication with the MCU.
A/D conversion reference voltage (REFP,
REFN) should be connected to load-cell’s supply
terminals. It can be connected directly to AVDD,
or through a resistor to reduce the power
consumption by the load-cell.

[94396103-Hx712-English-Datasheet.pdf](https://github.com/bogde/HX711/files/913104/94396103-Hx712-English-Datasheet.pdf)

tk5ep, 2017-04-11:
Thanks,

I have this doc, and i must be dumb, but can't understand for what this BAT
pin is good for...

I understand that it has an internal voltage divider and ADC of gain 1. If
a battery voltage is detected, then AVDD is used . If the pin is not used,
it still  AVDD !

Or what did i misunderstand ?

bogde, 2017-04-12:
@tk5ep the LT1761 looks awesome, but the store i usually get components from doesn't stock it. it's also kind of expensive and i'm afraid others trying to later build the board i'm trying to make will be put off by the cost.

would something like one of these work instead:
http://www.tme.eu/cz/Document/16637f822e97fb29c08daa0179c3e821/SPX3819-DTE.pdf
http://www.tme.eu/cz/Document/8266aa386d25c81d5f3a0a0fc53614a3/LDK120.pdf
http://www.tme.eu/cz/Document/d395f5cae42cb44e9cd4305f8b9d12b3/LDCL015.pdf
http://www.tme.eu/cz/Document/cf9e8eaf4cf901b3fb507741facf1eca/LD2985.pdf
http://www.tme.eu/cz/Document/8266aa386d25c81d5f3a0a0fc53614a3/LDK120.pdf

any help is highly appreciated!

tk5ep, 2017-04-12:
> 21/SPX3819-DTE.pdf

Hi !

@Bogde

I don't know these regulators.

The LDK120 is limited to 3.5V. Regarding all other parameters, the LD2985
looks good.

It has a good low output noise.
Only the quiescent current is "much" higher than the LT1761.

It depends on what you're looking for. But for your board, the LD2985 seems
to be a good choice.

Just my 2 cents,
Patrick

tk5ep, 2017-06-10:
Hi Bogde,

Any progress with your board ??
My hive scale prototype with the HX711 is now at work under a hive but would like to make a new one with the HX712 for more thermal stability.
[Here the realtime graphs](http://www.egloff.eu/apaguard/apaguard_graph.php) 

Cheers,

electrokean, 2017-06-10:
@tk5ep if you had another HX711 and set of load cells in an enclosure next to the hive with a known weight on it, you could potentially use that to work out the temperature compensation for your hive.  Just a thought, as the temperature variation may be partially in the load cells, not just the HX711.

electrokean, 2017-06-10:
BTW - nice work on the data collection & graphs

bogde, 2017-06-10:
@tk5ep unfortunately i didnt have the time to work on this.

flowmeter, 2017-06-27:
@tk5ep Page#2   14 BAT Analog Input Battery voltage detection input（connect to higher of AVDD or
DVDD when not used）
@bogde [LT1761](https://www.aliexpress.com/item/Free-shippin-10pcs-lot-LT1761-LT1761ES5-BYP-SOT23-5-Regulator-original-authentic/32524616582.html)

flowmeter, 2017-06-27:
Analog Input and Battery Detection Differential inputs are designed to interface directly with a bridge sensor’s differential outputs.
It can be programmed with a gain of 128 or 256.
When 5V reference is used, these gains correspond to a full-scale differential input voltage of ±20mV or ±10mV respectively.
Single-ended battery input signal (BAT) can be directly connected to a battery output (2.6~5.5V). On-chip 56kΩ and 3.2kΩ resistors divide this input voltage to provide suitable voltage to the ADC, which has a gain of 1. When detecting battery input, AVDD is used as the ADC’s data conversion reference voltage. When battery detection circuitry is not in used, a MOS switch is used to turn off the resistor divider path
current.

tk5ep, 2017-07-05:
@flowmeter 
Thanks, that is what is written in the datasheet... But practicaly ?
I can't figure out for what the BAT input is good for ??
AVDD is always used as the reference voltage through the internal MOS switch and via the VOUT pin !

For me, the datasheet is probably wrong... This input is for measuring the battery voltage through the internal multiplexer and PD_SCK pulse 26 is used to measure this voltage with a gain of 1. Not of great interest, except if you need to measure the supply voltage.

Just my 2 cents, Euro cents of course ;-)

tk5ep, 2017-07-05:
@electrokean 
Thanks for comment. I came to the conclusion that the HX711 thermal drift is much less than the one of the  cell... So, it wouldn't bring a lot to try to compensate it.
For the moment, i'm quite happy with the real world results.. Only complain is that i have some strange power consumption when using the HX711 under certain conditions. I must investigate this further.

electrokean, 2017-07-06:
@tk5ep are you using `scale.power_down()` and `scale.power_up()` between readings?  Worth doing if you are not reading at a fast rate.

tk5ep, 2017-07-06:
@electrokean 
Yes, i do. To save the max of power, i at first switched off all power on  VSUP, but that was a bad idea... It's better to leave it and use the power_down function. I do not use the internal voltage regulator and have an external 5V low noise regulator i'm switching off with the BASE pin. Otherwise, when AVDD is still applied, it consumes 20mA when VSUP is off !

tk5ep, 2017-09-27:
@bogde 
I finaly forced myself to find some time to finish my project and replace the HX711 with the HX712.
I soldered the chip on a test board but so far did not connect anything at the analog inputs.

I use the original untouched library for now. The software does seem to dialog with the chip as i have the VOUT voltage going high during the measure, but it does not read anything. I get a "Nan"

During your test, did you change anything during your own tests ?

Thanks,

tk5ep, 2017-09-27:
@bogde 
Hi, i was a little bit too in a hurry...
I didn't go through my calibration procedure and some calibration values were missing.

Now i can talk and read with the HX712.

I also modified the library to be able to use the additional amplifier values, but i'm not a good programmer and i'm not sure if i made it right.
At least does my modified library work with my original parameters.

Can i send you the library so you can check it ?

Patrick,

bogde, 2017-09-27:
yes, please send it over.

tk5ep, 2017-09-27:
Here it is. Couldn't drop it in your HX712 folder.
[HX712 master.zip](https://github.com/bogde/HX711/files/1338008/HX712.master.zip)

JamesKDouglas, 2019-04-08:
Regarding the BAT pin, it seems to me that the unit is comparing the battery voltage to the signal out from the load cell and compensating for the expected drop as the battery becomes depleted. The signal from the cell is a ratio of the input voltage, so the amplifier is able to consider the input voltage, do the calculation then report the weight? This ratio technique is common in analytical chemistry devices (that's my background).

I started adding a pullup resistor to the clock pin because the power down function in the library relies on the pin staying high. But during deep sleep many microcontrollers allow the pin to float, so the amplifier will turn on again by accident.