bogde/HX711 #41: HX711 Standalone

Priya003, 2016-07-17:
Hi, wanted to use HX711 24-bit adc standalone to measure voltage range of 2mv - 20mv.
Problem is:
1) I cannot interpret the adc output count.
2) What should i do in my code(code below) so that it prints voltages not decimal values.
Tried with different values of voltages(shared below), and i am totally confused with the readings. I don't know how to convert these values in voltage. What is the logic behind. Please help!
Readings:
0mv   --- 5219
1.1mv --- 5095
2.3mv ---  4981
2.5mv ---  4960
2.9mv ---  4918
5mv ---  4693
10.6mv ---  4075
15.1mv ---  3597
22.7mv ---  2700
40.7mv ---  562
5volts --- -8388608
-5volts --- -8388607
Board is arduino leonardo, and a simple voltage divider is used for voltage generation(for testing only). The voltages are measured and calculated from multimeter and hand calculations. I need HX711 do this for me. What is the differential ADC count and voltage relationship...?
A bogde/HX711 library is used: https://github.com/bogde/HX711
Code:
#include "HX711.h"
#define DOUT  3
#define CLK  2
HX711 scale(DOUT, CLK);
void setup() {
  Serial.begin(115200);
}
void loop() {
 long avg = scale.get_value(60);
  Serial.print("Digital Code = ");
  Serial.println(avg);
}

electrokean, 2016-09-23:
Don't know if you're still working on this, but essentially your offset is 5219 and your scale is somewhere in the range of -107 to -114.
You also shouldn't be connecting the input outside the specified full scale differential input range, typically +/-20mV or +/-40V (5V supply and gain of 128 or 64).
Also you can't connect a negative voltage at all, or you'll damage the chip.  The input is intended to be +/-40mV offset from AVDD/2.  This might be why your readings are changing so little over the 0-40mV range you tested with.

electrokean, 2016-12-05:
@killerkitties Can you provide a drawing of your wiring?  You mention a wheatstone bridge configuration, but if measuring a voltage you need to keep the circuits isolated with respect to ground.

The input of the HX711 is designed to measure +/-20mV or +/-40mV centered at about half the supply voltage (i.e. a bit below 2.5V in typical use).  Going very far outside of this range the internal amplifier could easily hit linearity issues.

There is some discussion very similar to what you're doing elsewhere which may help, and at least one mention of non-linearity...
http://electronics.stackexchange.com/questions/246418/hx711-voltage-measurements
https://forum.arduino.cc/index.php?topic=413099.0

bogde, 2017-12-06:
since there wasn't any feedback on this, i'm closing the issue. feel free to open it again if needed.

closed 2017-12-06 by bogde