bogde/HX711 #42: Default values for OFFSET and SCALE.

obiwan-bartek, 2016-07-30:
Hi,
I suggest to add a default values for OFFSET (as 0), and for SCALE (as 1), to be able to use .get_units() without first setting them in code.

bogde, 2016-09-23:
can you elaborate on that? using get_units for a SCALE of 1 won't make much sense i think...

ReDetection, 2016-10-02:
I can. Currently SCALE is 0 at start, so when you try to get units, it reports infinity, which is a result of anything divided by zero. This is a way less sense than have default scale of one 😉

bogde, 2016-10-02:
Good point, thanks for clarifying. I was thinking you wanted something that will give you grams or kilograms by default. I'll make the changes asap.

bogde, 2016-10-02:
This should be fixed in 38f8ff0bda29b2377a9a6d3fe3eb25690c05bbe2.

closed 2016-10-02 by bogde