bogde/HX711 #43: Incorrect result with Two loadcells

alexandra110, 2016-09-10:
Hi every one
When i connect two loadcells via junction box to hx711 A input i got some values that changes not in linear form.it means that when the load is more than a special value , hx711 sends lower values instead of higher ones.
Can every one help me?

electrokean, 2016-09-11:
This could be a software or hardware issue.
On the software side, the only thing that comes to mind is the discussion in this issue: https://github.com/bogde/HX711/issues/23
If that doesn't solve it, then you'll need to provide more info, like load cell details (type/model/capacity), approx. weight level that triggers the issue, a diagram and/or photo of your wiring, and your Arduino sketch

electrokean, 2016-09-11:
In reference to issue 23, I see there was a patch committed last year.  It is possible that this doesn't work as expected, so you could try reverting that.
See https://github.com/bogde/HX711/pull/17/commits/1f9aaee116ce9463bbc4a310beea4d24d0ff7a1b for the changes

alexandra110, 2016-09-11:
No one???

alexandra110, 2016-09-11:
Thanks for your help. How wiring can make these kind of errors?

alexandra110, 2016-09-11:
I read link you refered. it was very helpful. thanks for your help.
Base on this it is possible that i don't read hx711 output correctly , so i misunderstand that the output changes  is not linear. is it true?

electrokean, 2016-09-12:
Hi @alexandra110 
I see you're new here.  Welcome!  Please keep in mind that github is not a general support forum.  It is for reporting issues with the software provided.  We need to determine what your issue is, and whether it is a software issue, or something else.
So for anyone here to help you, you will need to provide a lot more detail on your setup and exact problem.  At this point we will just be guessing.

alexandra110, 2016-09-12:
Yes.i'm new here. Tanks. I didn't know that.
Ok i will come back with details

FabianInostroza, 2016-09-22:
Did you solved it?

alexandra110, 2016-09-23:
I couldn't spend time for this project any more. I will deal with that problem in coming week and share the result in this site.

FabianInostroza, 2016-09-23:
Try the changes I've made in my fork. I'm using similar code reading a single load cell in channel A and works fine.

alexandra110, 2016-09-23:
I also used single load cell with channel A, you are right, it works fine. But the problem appears whe l add two loadcell output then using channel A. I guess that it is hardware problems.i will check it.

EvertonCosmo, 2018-12-09:
@alexandra110 Did you solved it ? I have same problem

alexandra110, 2018-12-13:
unfortunately no... i didnt spend any more time for it