bogde/HX711 #44: Can't use library due to yield calls on Arduino

Ilidur, 2016-09-18:
I tried using your library as is but it seems like yield() is not a standard call in the Arduino IDE. Trying to google it has lead me to install a wrong Scheduling library and couldn't find a right one either. In the end I had to remove the two calls before installing it.

electrokean, 2016-09-18:
If you have any recent version of Arduion it will define yield().  It was added in Arduino 1.5.1 in late 2012.
For the AVR platform it is an empty function in hooks.c, but can be overridden on other platforms which use cooperative multi-tasking.
What version of Arduino are you using, and what board are you compiling for?

Ilidur, 2016-09-18:
Ah I see. The arduino package in ubuntu is stuck at 1.0.6. The "2:" in front confused me and thought it was a higher version.

closed 2016-09-18 by Ilidur