bogde/HX711 #45: tare function for two load cells

piflixe, 2016-09-23:
I am trying to use the library with two load cells connected to one HX711 (channel A and B).
As described in the README I can switch between the different load cells with the set_gain function.

How can I use different set_scale and tare for the different channels?

piflixe, 2016-09-23:
that's exactly what I tried but it seems there is only one tare value implemented. After tare() for both, it uses the tare value of channel B for both channels.

piflixe, 2016-09-23:
I totally understand you can't fix this at the moment. I was just wondering if I was just too stupid to use the library properly. So this main question was answered quickly. Thanks for that!

I am using a workaround now manually storing the tare value for channel B. It is ok and working for now. Feel free to use the idea and usecase for further improvements.

bogde, 2016-09-23:
what i meant was you _could_ (not sure) call tare() again; but depending on your use case that may not be helpful.

closed 2016-11-15 by bogde