bogde/HX711 #47: Current master does not work

ReDetection, 2016-10-02:
I spend almost a day trying to figure out why I receive random readings. I checked wiring several times, each pair of load cells, etc, until I tried another HX711 library which worked great but does not have a powerdown mode and so on. 

I already figured out it's because of pull request #38 (commit 0c444e2 works fine), so probably every issue since that pull request may be related, but I would like to explicitly leave it here until the issue is resolved, so nobody going to spend a lot time in headaches.

bogde, 2016-10-02:
I just reverted the changes introduced by the pull request #38. Thanks for your input and I apologize for wasting your time. If you could test the current master and confirm it's working that would be great. Thanks again!

ReDetection, 2016-10-02:
Yep, I confirm master branch is working again. Guess we can close the issue.
[image]

And thanks for implementation of powerdown! 😴

closed 2016-10-02 by ReDetection