bogde/HX711 #50: Setting GAIN after each read

eghantous82, 2016-10-20:
After reading the datasheet for the Hx711 it appears that that 25-27 pulses on PD_SCK are required to set the GAIN and INPUT.  Looking at the code, it looks like it's looping 1-3 times.  Am I missing something?

PD_SCK pulses | INPUT | GAIN
25                          A        128
26                          B         32
27                          A         64

bogde, 2016-10-20:
shiftIn (x3) should pulse the clock 24 times; then the code i think you're referring to pulse it again 1, 2, or 3 times depending on the required gain factor, that should give 25, 26, 27 pulse, like in the datasheet. let me know if you think that may not be correct.

eghantous82, 2016-10-21:
Thanks, sorry I missed that.

closed 2016-10-21 by eghantous82