bogde/HX711 #51: HX711 Drift

OO999, 2016-11-21:
Hi, 
I'm using an HX711 amplifier with  50Kg 4 load cells, like this:
https://learn.sparkfun.com/tutorials/load-cell-amplifier-hx711-breakout-hookup-guide

My problem is that the reads is drifting, even with no load on scale.

Board: ESP8266 e-12
Power source: MB102 Breadboard Power Supply Module 3.3V 5V For Arduino

I'm using Bogde's Hx711 library (https://github.com/bogde/HX711).
My
[image]
 code is very basic:

void loop() { 
  if(scale.is_ready()){
   Serial.print(",");
  Serial.print(scale.read_average());  
  }else{
     Serial.println("scale not ready");  
  }

  delay(1000); 
}

And the output is keep drifting up (see attachment).

Any ideas what cause the drift? and how do I fix it?

Thanks!

electrokean, 2016-11-21:
Drift is pretty normal with load cells, due to temperature and mechanical settling, but that seems excessive.
How are you supplying power to the MB102?  USB or plug pack?
Can you measure the load cell excitation voltage with a multi meter and see if it is stable over time?  Such as over 10 minutes, and see if it is dropping in correlation to the drift.

electrokean, 2016-11-21:
Actually, I say that seems excessive because you're showing a movement of just under 4% (2500/67000), but I've got nothing really to go against.  You're not tareing or scaling the readings.
That change may only equate to a few grams of drift over your 200kg (4x50kg) or range.
Please add some code to tare then scale the platform, and then you can quantify the drift.

OO999, 2016-11-21:
Thanks for the quick replay.

This is the plot for about an hour read, with 20Kg on.
I newbie with this kind of things, but why do the the raw reads has to drifts so much (even if it just 4% out of 200Kg)?

For my understanding , if all parameters are fixed - load, power source, temperature, code side - isn't the read suppose to be basically the same over time (lets say an hour)?

[image]

electrokean, 2016-11-21:
I suspect you have a faulty load cell or incorrect wiring.  That works out at over 0.5kg drift.  Not as bad as 4% of 200kg, but still awful.
Are these 3 wire or 4/6 wire load cells?  If 4/6 wire, you can test each individually to try isolate the issue.
If they are 3 wire, well... I'd suggest reviewing your wiring and then getting a decent load cell so you don't waste your time.  ;)

electrokean, 2016-11-21:
BTW, you didn't really try any of my initial suggestions.  Do you have a multimeter?

OO999, 2016-11-21:
Its a 3 wires load cells. and I do have a multimeter, what kind of tests can I do to isolate the problem?

Thanks for you help electrokean!

OO999, 2016-11-21:
I'm not sure its relevant, but after 3 hours with the same load on scale,
it look like the drifts decrease to almost a flat line, as seen below:
[image]

electrokean, 2016-11-21:
As I suspect a wiring issue, I recommend starting there.
Have you confirmed the load cell wire colours according to the Sparkfun page you linked?
Are you using the Sparkfun HX711 and Load Combinator PCB?
Can you provide a drawing of all your wiring, and maybe a photo?

The ESP8266 runs from 3.3V, but the HX711 can run from 2.4V to 5.5V.  What voltage are you running the HX711 at?
This voltage will determine the excitation voltage it can supply to the load cells.  Normally these pins will be labelled E+ and E- for the load cell connections.
Can you measure the voltage between E+ and E-, and also see if it is stable over time?

electrokean, 2016-11-21:
OK, just confirm that you've got the 3 wire load cells wired correctly - the correct wiring configuration isn't obvious.  See http://electronics.stackexchange.com/a/196064 for another wiring example that I think is clearer.

I've only used the HX711 at 5V.  Running it at 3.3V should be OK, but may make it more succeptible to noise.  I actually stopped using the HX711 and went to another chip for my own design, but this library was really helpful to me when I first started.

You'll get better results if you limit your combined load cells to a bit above your maximum, otherwise you're wasting the range of the ADC.  But don't get too close as the load cells can become non-linear close to their limits.  The 4x50kg load cells are typically used in cheap scales for weighing people.

The 3 wire load cells will work best in a set of four, but you can wire them up as single or doubles.
See http://www.allaboutcircuits.com/textbook/direct-current/chpt-9/strain-gauges/ for examples.

I now always use a 6 wire "beam style" load cell when I want accurate measurements, or 4 wire for testing.  I've have actually never used a 3 wire load cell on my own projects, but I have helped rewire them for friends.

OO999, 2016-11-21:
Thanks electrokean, I will take a look on the beam style load cell

closed 2017-01-11 by bogde

GeorgeHahn, 2017-05-18:
@electrokean Curious - what chip did you switch to for your project? How do you like it compared to the HX711? Cheers!

morphogencc, 2017-05-18:
I'm seeing the exact same issue listed here -- I bought four of [this](http://www.mouser.com/ProductDetail/Measurement-Specialties/FX19-0-1-0001-0010-L/?qs=sGAEpiMZZMs29kr3d%252bndIyR7UOFvDGdDpbVFUu47pvc%3d) load cell, and wired them up with a Sparkfun Combinator and a Sparkfun HX711 Breakout board.

I wired them up so that red (supply +) goes to '+', yellow (output +) goes to 'C', and black (supply -) goes to '-'.  The white wire (output -) is current left unattached.

The HX711 is outputting a steady 4.35 Volts, so it doesn't seem to be a power issue.  I've tried adding 0.1 uF (for power noise) and a 10 nF (for RF suppression) to the sensor leads, but this doesn't seem to be the issue.

I'm under the impression this is the correct wiring for the part; if so, then the HX711 chip itself or library are the only things I can imagine that'd cause this sort of dramatic drift in the readings.

electrokean, 2017-05-18:
@GeorgeHahn The part I use is the Texas Instruments ADS1232.  http://www.ti.com/product/ADS1232
This is very similar to the HX711 but much better documented and well supported (and of course more expensive!).  I had problems with the HX711 in getting consistent readings at a high data rate.  I would regularly get bogus readings, which were painful to filter out.  I presume this was the HX711 shift register getting out of sync.  Never had any such problems with the ADS1232.

I have my own PCB design that combines the ADS1232, ATmega328, temperature sensor, plus power and comms circuits.  It runs an Arduino sketch, and connects back to a control system/data logger via RS232 and a 12V supply, but is pretty much based on the ADS1232 reference design.  It is used in the field for automatic weighing of penguins and wallabies here in Australia, and was designed to be a drop in replacement for a 20 year old design that had been used in Antarctica.  It can return weight data at 50 readings per second (100 SPS averaged in pairs).  To get the 100 SPS, I use an external crystal.  I also developed extended firmware for newer projects that supports additional commands, including measuring noise levels.  I've had these running for a couple of years now in pretty harsh environments, and the only problem has been with rats eating cables!

electrokean, 2017-05-18:
@morphogencc you don't want to use the Sparkfun combiner PCB if you have 4-wire load cells.  It is only intended to make wiring easier for the 3-wire half-bridge load cells.
If you have multiple 4-wire load cells, then just wire all 4 wires of the same colour together.  See the datasheet on the Mouser page for details of the wiring of that load cell.
Maybe try with just one load cell at a time first to check they all give consistent results, before paralleling them.  You'll obviously need to recalibrate once combined as you'll have 4 times the capacity.

GeorgeHahn, 2017-05-23:
@electrokean That sounds like a great project! Is your extended firmware open source? I'd love to take a look at it and port it to Simblee (which uses a Nordic ARM part).

electrokean, 2017-05-26:
@GeorgeHahn It isn't currently open source, but I'll think about it.  Don't see any issues, but I just need to spend a little time on it before a public release, and I don't have spare time right now.

GeorgeHahn, 2017-05-26:
@electrokean Let me know if you'd like help with tidying! I'm just about to order prototypes that use the ADS1232 part and will have a good bit of time to spend working on firmware.

electrokean, 2017-05-28:
@GeorgeHahn sure, shoot me an email.  Contact details can be found via website listed on my profile.

flowmeter, 2017-06-27:
@electrokean which library do you use with ADS1232 ? What is your advice for a 6 wire "beam style" load cell provider ?

electrokean, 2017-06-27:
@flowmeter It is my own code, based on info from the TI datasheet and app notes.  @GeorgeHahn has a copy of my code and we are working to release something.
The beam style load cell will depend a lot on your requirements.  I typically use ones that are typically several hundreds of dollars, and that probably wont suit most people.  I've sourced off the shelf platforms (e.g. Ohaus) and also built my own out of stainless steel with multiple PT Global load cells in parallel.  I've also bought cheap Chinese beam-style load cells from AliExpress, but for me those are only for playing around :D

evandavey, 2017-12-12:
@electrokean, @GeorgeHahn - any progress on a firmware release?

AGenchev, 2018-08-31:
@OO999 I play with HX711 and I get also big drift. In a month I'll try wiring a different ADC (12-bit) combined with an amplifier giving out 0..3.3V for load 0..4kg and see what happens then.

davidjonz2005, 2019-04-13:
@electrokean I am working on a project in weighting but w/o any human involved and I managed to read my HX711 with a stm32f030. I have a consistent reading in weight near 25 Kg but when I weight my self It gives me different value in each read. It is not correcting even by heavy average such as 100 sample. I want to know have you had such this problem in your project and what was your procedure to solve it

GeorgeHahn, 2019-04-14:
Is your load cell rated for large weights?

davidjonz2005, 2019-04-15:
@ I bought a commercial 180 kg weight scale and removed the board so I replace it with my hx711 module with remaining 4 loadcell

electrokean, 2019-04-15:
@davidjonz2005 this issue has been closed for over 2 years!
I suggest you start a new issue, and provide a lot more info on what you have and what you are seeing, and what troubleshooting steps you've already taken.
I don't have any free time at the moment so I am not actively monitoring issues here (especially as I no longer use HX711), but I suspect others may be able to help if you provide enough relevant info.
I often find that just carefully writing up the details on what you are doing and seeing will help you diagnose the issue yourself.

toddmo, 2024-02-02:
Mine's doing the same thing. Well, it's slowly drifting up and down. I suspect it's undulations in the mains voltage because nothing else in the setup is changing.

AGenchev, 2024-02-05:
The load cells are highly sensitive to the temperature. How sensitive ? Individually - we've seen positive, negative. We've tested with TI made ADC and it wasn't much better. At the end we abandoned the project (it was a hobby).