bogde/HX711 #56: Gain explanation

brownd407, 2016-12-17:
Hello all,

First thanks a lot for your code. I ordered a load cell and i almost managed to make it work.

I have a very newbie and stupid question but i was not able to find the answer.
What exactly is / does gain parameter on xh711? I mean what is difference between 128 and 64 ?

Thank you in advance

bogde, 2016-12-17:
from the datasheet:

Channel A differential input is designed to
interface directly with a bridge sensor’s
differential output. It can be programmed with a
gain of 128 or 64. The large gains are needed to
accommodate the small output signal from the
sensor. When 5V supply is used at the AVDD pin,
these gains correspond to a full-scale differential
input voltage of ±20mV or ±40mV respectively.
Channel B differential input has a fixed gain of
32. The full-scale input voltage range is ±80mV,
when 5V supply is used at the AVDD pin. 

hope this helps.

closed 2017-01-11 by bogde

Ballen7, 2018-03-26:
I am new to electronics. Does this mean that voltage increments below ±20mV wont be detected?