bogde/HX711 #57: Arduino compilation error: OFFSET & SCALE initialisation

freekeys, 2017-01-03:
Hello, 

Apologies in advance if this is a noob question and/or an library installing error.

I've installed the library to the Arduino IDE (1.0.5-r2 on Windows 7) and am trying to compile the code in the examples/HX711SerialBegin file.

I keep getting the error ..

> In file included from HX711SerialBegin_lmedit.ino:1:
C:\....\Arduino\libraries\HX711/HX711.h:16: error: ISO C++ forbids initialization of member 'OFFSET'
C:\....\Arduino\libraries\HX711/HX711.h:16: error: making 'OFFSET' static
C:\....\Arduino\libraries\HX711/HX711.h:16: error: ISO C++ forbids in-class initialization of non-const static member 'OFFSET'
C:\....\Arduino\libraries\HX711/HX711.h:17: error: ISO C++ forbids initialization of member 'SCALE'
C:\....\Arduino\libraries\HX711/HX711.h:17: error: making 'SCALE' static
C:\....\Arduino\libraries\HX711/HX711.h:17: error: ISO C++ forbids in-class initialization of non-const static member 'SCALE'

I've done some googling and there are some comments suggesting that the variables shouldn't be initialised to a value in the header file but have tried removing these with no luck.

Any help would be much appreciated.

Thanks

bogde, 2017-01-03:
is there a way you can use a newer version of the arduino ide?

freekeys, 2017-01-03:
Thanks for the quick response. I realised myself just before you posted how out of date my ide is - haven't been developing for a while on this laptop.. I'm installing now so will let you know if this sorts it. Thanks.

freekeys, 2017-01-03:
Sorted! Stupid.. thank you

closed 2017-01-03 by freekeys

SFS87, 2019-03-10:
I am having this exact same issue. What version of the IDE was used?

bogde, 2019-03-10:
what version are you using?

SFS87, 2019-03-10:
Thanks for replying so quickly! I initially used the same version as OP, 1.0.5  and have tried a couple newer versions, and the newer versions give me additional errors. I tried the latest 1.8.9 and 1.6.13 and neither worked.

bogde, 2019-03-13:
can you post the error messages you're getting? also, did you select the correct board?