bogde/HX711 #58: Hi bogde

Hariskhalil, 2017-01-04:
i have old tensile testing machine which has been expired, i thought to make that machine for tensile testing using machine's own load cell and mechanical assembly.
i figured out that i need avia semiconductor 24bit adc, arduino, for calibration of load cell( 0.1 mV / V) and your library same power supply.
i am using your library, and i have been working for last 5,6 days and i have frustrated because of no result. 
The issue, i am facing is my load  showing unusual behavior when ever i connect load cell and run basic program(yours) it shows continuously fluctuation values with out even little accuracy.
i have tried a lot with different ways (basic program, changed programmed, etc etc)
Kindly help me to solve issue  
[image]

above is status pic

bogde, 2017-01-11:
what happens if you add a delay between readings?

Hariskhalil, 2017-01-11:
same thing happens , same behavior  
Do u think its load cell issue ? i have also attached  voltage divider 0 to 20 mV using arduino 5V supply to it just for seeing result but i got same behavior .

electrokean, 2017-01-11:
I suspect a wiring issue or a damaged chip.

You say you attached a voltage divider - how did you do that exactly?
The HX711 is designed to measure a voltage difference of +/-80mV max between the two inputs when at a potential near half the supply voltage.  It isn't designed to measure voltages close to either of the supply rails.  The datasheet says they must be kept above 1.2V, and below the supply less 1.3V.

Hariskhalil, 2017-01-11:
i used simple voltage divider circuit consist of one resistor & one V.R in series supplied by arduino  5 V supply and its output (approx 0 mV to 22 mV) gave into arduino inputs pins for seeing response after using load cell.

electrokean, 2017-01-11:
As I mentioned that wont work.
The inputs are differential and require voltages centred at around half your supply voltage, with +/-80mV max differential between the IN+ and IN-.
So for example with a 5V supply INA+ may be 2.501V and INA- 2.499V to get +2mV differential voltage.
Voltages at the inputs less than 1.2V or greater than 1.3V below the supply are outside device specification.

Hariskhalil, 2017-01-27:
As far as i can understand that you are saying by interfacing voltage divider output signal directly to HX711 module input will not work because it gives only some mV (checked using voltmeter) signal & does not give half of input supply volts to give differential mV signal ?

i had my load cell connected in same manner as in pic 
[image]
it showed same behavior as voltage divider. 

should i change cell or add some circuitry before connecting to Hx711?
if i change cell then what specs should load cell to work properly with Hx711 ?

electrokean, 2017-02-14:
That wiring looks fine.  It may be that your HX711 board is faulty.  Do you have another one to try?
This is the style of load cell I recommend https://www.sparkfun.com/products/13329
You want a 4 or 6 wire load cell (not 3 wire) and obviously rated to suit your needs.

closed 2017-03-21 by bogde