bogde/HX711 #60: Calibration issues - erratic values from demo sketch

freekeys, 2017-01-14:
Hello,

I've set up my load cells and tried the calibration process and HX711SerialBegin code but am having trouble.

I am using the Arduino IDE (1.8.0) on Windows 7 with an Arduino Uno. I have 4 load cells wired together into a Sparkfun Load Cell Combinator and then into a red (not Sparkfun) HX711.

Here are links to the components:
**Load cells** - http://www.ebay.com/itm/261432803492
**HX711** - http://www.ebay.com/itm/181357821017
**Combinator** - https://www.amazon.co.uk/dp/B01LXUNNPL/ref=pe_1909131_77697121_tnp_email_TE_AMZLdp_3

When I run the raw (no changes) HX711SerialBegin code from the library the Serial Monitor prints out very steady values around 0 - but these figures don't seem to change when I add weight to the sensor (see serial readout below).

[image]

I then ran some code as per the calibration instructions which seems quite variable in what value it gives me, generally resulting in a value between -0.5 and 0.5 which seems low.

**Calibration Code**
```
#include "HX711.h"

HX711 scale;

void setup() {

  // parameter "gain" is ommited; the default value 128 is used by the library
  // HX711.DOUT  - pin #A1
  // HX711.PD_SCK - pin #A0
  scale.begin(A1, A0);

  
  Serial.begin(38400);
  Serial.println("HX711 Calibration");

  scale.set_scale();
  scale.tare();

  Serial.println("You have 15 secs to place a known weight on the sensor");

  delay(15000);

  Serial.print(scale.get_units(5)/380); // known weight is 380g
 
}

void loop() {
}
```

**Value from calibration code in HX711SerialBegin sketch**
```
scale.set_scale(-0.29);                      // REPLACES 2280.f ...this value is obtained by calibrating the scale with known weights; see the README for details

  scale.tare();				        // reset the scale to 0
```

This then results in very variable results and fluctuating numbers in the demo sketch.

[image]

Any help would be much appreciated! Thank you :)

electrokean, 2017-01-15:
You have a total load cell full scale weight of 200kg (4 x 50kg), and you're trying to calibrate with a 380g weight?  That is less than 0.2% of full scale.
If you look at the specifications of those cheap half-bridge load cells, they can have creep and zero drift of 0.1% of full scale per minute.  And they can also have non-linearity and repeatability of 0.05% FS.
I suggest trying to calibrate with a much bigger known weight, such as a large bucket of water.
If that doesn't work, then you probably have a wiring problem.
If you want decent accuracy for smaller weights, you need a much better (4 or 6 wire) load cell.

bogde, 2017-12-06:
since there wasn't any feedback on this, i'm closing the issue. feel free to open it again if needed.

closed 2017-12-06 by bogde