bogde/HX711 #64: How to modify the code in order to use the built in clk rather from the arduino.

Fieryfire, 2017-02-21:
How to modify the code in order to use the built in clk rather from the arduino. So that i can free one slot?

electrokean, 2017-02-22:
You have to have both the clock and data pins connected, otherwise there is no way to read the data.
In "built in clk" of the HX711 is purely used for conversion timing, not communications.

Fieryfire, 2017-02-22:
Hi Kean,

Thank you for the reply. I also have another question. Is it possible to
connect a quarter bridge to Hx711 rather than connecting a full bridge?

Thanks,
Vivin

electrokean, 2017-02-22:
You must use a full bridge with the HX71 inputs1, but you can probably get away with building the rest of the bridge being made from fixed value resistors.  It won't be as good as a quality load cell with four strain elements, but should work.
Do a google search for "load cell quarter bridge" and you should see what I mean - especially under images.
Use good resistors, 1% or better with a good temperature coefficient if you expect large temperature variations.

Fieryfire, 2017-02-23:
Basically what i am trying to do is sense about 30 strain sensors in
quarter bridge. i think this would require me to have about 30 hx711 as
well? and 30 quarter bridge? so are you sure that i would in fact be able
to get a voltage reading from the sensors?

All i want to do is get the strain value from 30 strain sensors, and get it
into an arduino. and something that requires simpler circuits like quarter
bridge with an amp to get the arduino convert to a digital format

electrokean, 2017-02-26:
You will need to test it with your particular strain gauge, and see if you can get the accuracy that you require, and if your skills are up to the task.
Yes, you'll need 30 HX711 boards - they have two input channels (A&B), but the B channel has limitations (gain & constant swapping) and many people have had problems getting them to work.  If you need that many, then it might be worth spending some time to see if the B channel works for your needs.
With careful programming, you can interface to multiple HX711 from a single Arduino - but I'd suggest no more than maybe 5.  Search the closed issues as there was some discussion on this in the past.

Fieryfire, 2017-02-27:
Wouldnt it be that the Hx711 must be interfaced to arduino. Hx711 takes in
the analog inputs. amplifies , converts to digital data, and the arduino
reads it and puts it over a serial port. Is that a possibility? and i
probably wouldnt be needing so much accuracy, as i am compensating that
with an algorithm that takes in data from all 32 sensors.

electrokean, 2017-02-27:
Yes, isn't that what I said?

closed 2017-03-21 by bogde