bogde/HX711 #65: Code will only return 0.0 for everything

LewisAstbury, 2017-02-22:
Hi i am using your SerialBegin example and i can't seam to get the load cell to return any values. I am using an Intel Edison, stacked on a Sparkfun Base Block, then GPIO Block, this is wired over to a Sparkfun Load Cell Amp which is wired to a Sparkfun 200kg Load Cell. I have the right pins and i am able to compile the code and run it, but in the serial monitor it just displays 0.0 for every stage, there must be something i am doing wrong but i can't quite figure it out. Do you have any suggestions to why this is happening? 

Thanks, Lewis.

bogde, 2017-03-21:
sorry for the delay, did you fix the issue yet? are you sure your wiring is correct, especially the correct wiring of the clock and data pins?

morphogencc, 2017-05-18:
In my experience, if you're getting 0.0 regardless of the input, then you've miswired it somehow.  It took me several tries to get the wiring right, but once I did I got much more sensible outputs.

jrleeman, 2017-06-12:
@LewisAstbury As others have suggested, try swapping your clk/dat lines and ensure they are connected to the right pins on your dev board.