bogde/HX711 #67: NodeMCU continuously rebooting using HX711

AGordiGuerrero, 2017-03-15:
Hi guys,

many thanks for the nice work with this library. I am enjoying it and doing some nice scales.

I was using the library without problems in a NodeMCU 1.0 (ESP 12-E) but I changed to the smaller version of the board (ESP 12-F):
http://www.banggood.com/Mini-NodeMCU-ESP8266-WIFI-Development-Board-Based-On-ESP-12F-p-1054209.html?rmmds=search

With the same code, as soon as I include the line:

`HX711 scale(D6, D8) // or whatever pins I choose.`

it restarts over and over again and send the message via serie (115200 bd):

> ets Jan  8 2013,rst cause:2, boot mode:(3,6)
> 
> load 0x4010f000, len 1384, room 16 
> tail 8
> chksum 0x2d
> csum 0x2d
> v60000318
> ~ld
> 

After looking in several forums, I have not found the answer. Can anybody give me a clue?

Thanks in advance!

AGordiGuerrero, 2017-03-15:
First self-correction: 
Now it neither work with the NodeMCU 12E.

Maybe it is related with the version of the compiler (it is running under Windows, Arduino IDE 1.6.12). Extremely slow compiler since last update. I will keep looking.

bogde, 2017-03-16:
can you post your sketch? i'm trying the HX711SerialBegin demo (MacOS, Arduino 1.8.1) on a "D1 mini WeMos.cc" and seems to be working after cleaning up the setup section.

bogde, 2017-03-16:
i just tried v1.6.9 on windows and it seems to work too, again, after removing most of the test code from the setup section. i'm going to update to 1.8.1 on windows too and give it one more try.

AGordiGuerrero, 2017-03-17:
Many thanks for the answer.

I have tried the Arduino IDE V1.8.1, V1.6.9, V1.6.6 and V1.6.0. All under ubuntu 64 bits. No solution found, same error and reboot. I even reinstalled every library deleting the /home/user/Arduino directory.

I paste below the code. It includes lots of comments, some timing variable and debugging flags. Please tell me if it is not clear and I will clean it. In line 93 is the scale initialization which causes the reboot.

I also tested that it is not a power issue, powering from a bench DC power supply. I see a peak on current just in the reboot, but I supose it is normal.

```
// WEIGHT MEASURE AND I2C OLED 128X64 SCREEN
//---------------------******--------------------------//
//
// Based on:
//    - NodeMCU and Arduino open hardware.
//    - Adafruit and Bogde libraries.
//          https://github.com/bogde/HX711
//    - Sparkfun HX711 load cell sketches
//
//    - Arduino online open community. Thank you all!
//
// Licensed under Creative Commons (SH-BY-NC). @AGordiGuerrero
//
// Antonio Gordillo Guerrero
// anto@unex.es
// @AGordiGuerrero
// Smart Open Lab. Escuela Politécnica de Cáceres. Universidad de Extremadura.
// @SolEpcc 
//
//--------------------******---------------------------//

/*/ Version Control

  
  2017/01/02: V1.4 Update to use the mini NodeMCU ESP-12F module alone.
  2017/01/02: Measure loop duration marking the start and end of the loop.
  2017/01/01: Include timing measures inside the main loop in #DEBUG .
  2016/12/31: Implementation of tare from a button. Removing shift from 1g.
  2016/12/30: AGG start coding from SmartScarecrow code. Simplifies all the code to eliminate configuration of alarms

*/

/* How to calibrate the scale (from https://github.com/bogde/HX711)
  - Call set_scale() with no parameter.
  - Call tare() with no parameter.
  - Place a known weight on the scale and call get_units(10).
  - Divide the result in step 3 to your known weight. You should get about the parameter you need to pass to set_scale.
  - Adjust the parameter in step 4 until you get an accurate reading.
*/

#include <SPI.h>
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
#include "HX711.h"

// Mode debug. Send all the information via serial port. Includes timing variables.
#define DEBUG

// Defines for the OLED I2C display
#define OLED_RESET LED_BUILTIN  //4
#define SSD1306_LCDHEIGHT  64

// Language settings for the display. Default is English.
//#define ESP // Comment for English version

// Arduino NodeMCU pin mapping
#define D0 16
#define D1 5 // I2C Bus SCL (clock)
#define D2 4 // I2C Bus SDA (data)
#define D3 0
[... 310 lines cut ...]
  display.clearDisplay();

}

#ifdef DEBUG

// Print information to serial port
void print_to_serial() {

    //Current time
    Serial.println();
    Serial.println();
    Serial.print("Main loop iteration: "); Serial.println(iloop, DEC);
    Serial.println();
    Serial.print("     Weight ("); Serial.print(nmeasures);Serial.print(" measures): ");  Serial.print(weight, DEC); Serial.println(" g.");
    Serial.println();
    Serial.println("     Timing: "); 
    Serial.print("          Current time: "); Serial.print(millis(), DEC); Serial.println(" ms.");  
    Serial.print("          Last loop duration (without serial): "); Serial.print(loop_duration_beforeserial, DEC); Serial.println(" ms.");
    Serial.print("          Last loop duration: "); Serial.print(loop_duration, DEC); Serial.println(" ms.");
    
}
#endif

```

bogde, 2017-03-18:
i dont seem to be able to reproduce you're issue, but i'm not sure we're using the same boards. i removed your oled display calls and replaced them with console output and everything seems to work as expected.

these are the boards i tested with:

[image]
[image]

fishersolutions, 2017-03-20:
Hi All,
I have been having the same issue with the HX creating reboots on the ESP8266.
It only happens when I added the OLED library.

This code is is being executed and therefore yield() no longer works.
Once I commented out this code everything started working :) 

> #if ARDUINO_VERSION <= 106
> 	// "yield" is not implemented as noop in older Arduino Core releases, so let's define it.
> 	// See also: https://stackoverflow.com/questions/34497758/what-is-the-secret-of-the-arduino 
> void yield(void) {};
> #endif

The long HX711::read() is also will lock the processor forever if the HX711 is never ready.

AGordiGuerrero, 2017-03-20:
Many thanks for your attention!

I use basically the same hardware than you. NodeMCU and "micro" NodeMCU. All based on ESP8266 V1.0. I got several NodeMCUs and I get the same errors in them.

[image]

Still no luck. I did not work too much last days but tonight I still do not get the answer.

I do not understand the solution from Fishersolutions. Where I should comment the yield() function? Thanks!

I will keep trying. I think it is related with the wdt. There are a lot of forums with the same issue but I still not found the solution. :-(

bogde, 2017-03-21:
if i were you, i would first isolate the issue by commenting out the oled display code, to test @fishersolutions tip. btw, thanks David for chiming in. 

or you can try commenting out:

void yield(void) {}; 

in HX711.cpp and see how that works.

AGordiGuerrero, 2017-03-21:
Good advices. It worked!! :-) But still partially...

I commented  in HX711.cpp the lines:
`
/*  	// wait for the chip to become ready
  	while (!is_ready()) {
		// Will do nothing on Arduino but prevent resets of ESP8266 (Watchdog Issue)
	  		yield();
  	}
  */`
`
And the other yield() calll in the read_average function. It now does not reboot automatically. Good!

But the measure I obtain is continuously zero. In fact when I upload the HX711serial test code, I get the serial messages:
`Before setting up the scale:
read: 		-1
read average: 		-1
get value: 		-1.00
get units: 		-1.0
After setting up the scale:
read: 		-1
read average: 		-1
get value: 		0.00
get units: 		0.0
Readings:
one reading:	0.0	| average:	0.0
`
So the read seems to be -1 at start. Maybe it is due to the comment on the yield() function and the fact that the chip is not ready to measure?

I tested with different pins, different loads and different HX711 boards. I should construct the same with simple a Arduino for testing..... It is quite confusing (for a noob as me) to work with the different execution layers in ESP8266.

Thanks again!

bogde, 2017-03-21:
you shouldnt comment out the is_ready code. you probably get 0.0 readings because the chip is indeed not ready. according to @fishersolutions tip, you should comment out the definition of yield. you should see that in HX711.cpp, in your arduino libraries folder.

however, since the sketch seems to work without the is_ready code, but you dont get correct readings, that could mean your wiring is wrong, or something else is wrong with your hx711 board. is_ready should definitely return true if the communication is working. you might want to further investigate that with a more simple sketch.

dhmsjs, 2017-03-22:
Isn't the normal state of DOUT always high, until the HX711 has a reading available? So then without the is_ready() wait loop in place, any call to read() during that (100msec) not-ready time will return all 1's, which is the bit pattern for -1.  That is exactly what you are observing.

If you are getting a watchdog reset when the is_ready() wait loop is not commented out, then that says that is_ready() is waiting too long for the watchdog timer.  The purpose of having the "real" yield() in that wait loop is to kick (reset) the watchdog while waiting.  As @bogde says you can't just cut out is_ready(), so one way or another you have to make sure that the watchdog timer gets kicked while you are in that wait loop. 

From @fishersolutions' post it sounds to me like without OLED, the watchdog was getting kicked often enough even with the do-nothing yield() call in the wait loop.  Adding OLED caused enough delay (when added to the is_ready() wait loop delay) to time out the watchdog.  His fix commented out the do-nothing yield() which then allowed is_ready() to call the "real" yield() -- i.e. the one that really does kick the watchdog timer.  You need to do the same.

AGordiGuerrero, 2017-03-23:
Exactly! @fishersolutions gave the answer. Just commenting what he said solved the error. If not commented, watchdog is not feed in time and reboot occurs.

My fault was mixed with a wiring error (.......) that I found testing with a clone of the hardware.

Now I got problems with stability of the measures but it is for sure an algorithm issue.

Thank you all again! Cheers from Spain.

dhmsjs, 2017-03-23:
Good to know; thanks for the update!  

If you are having problems with random strange readings from the HX711, note that @bogde's library will do this if your system uses interrupts and those interrupts ever take longer than about 60 uSec to service.  In that case, if one of these interrupts occurs during the time read() is clocking data out of the HX711, it can cause the HX711 device to go into shutdown mode, which corrupts the rest of that reading (all subsequent bits will read out as 1s).  One possible fix for this problem is to prevent interrupts from occurring during the entire clocking sequence.  See my pull request for one version of this solution.

Poor connections on the HX711 wiring can also cause random strange readings, so make sure your connections are all good and tight.

zinkkrysty, 2017-07-09:
I have the same error on a NodeMCU v1. 

Managed to copy the entire code of HX711 into my program and it worked - got reasonable readings.

However, whenever I import the HX711 library and use it my code, it restarts the device

electrokean, 2017-07-09:
@zinkkrysty As per suggestions above, comment out or remove the section in HX711.cpp that says:
`void yield(void) {};`
and see if that works.
This will cause problems on any ESP based hardware, as the ARDUINO_VERSION isn't set as expected.
See also discussion in issue https://github.com/bogde/HX711/issues/73

zinkkrysty, 2017-07-09:
Tried that. Not working.

Actually I can reproduce the error with this code:

```
#include "HX711.h"

HX711 scale(D5, D6); // This line is creating the error, if I remove it, everything works

void setup() {
  // put your setup code here, to run once:
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(115200);
  Serial.println("HX711 Demo");
}

void loop() {
  // put your main code here, to run repeatedly:
  digitalWrite(LED_BUILTIN, LOW);
  delay(2000);
  digitalWrite(LED_BUILTIN, HIGH);
  delay(2000);
}
```

Where I'm not using the library at all besides initializing the instance

zinkkrysty, 2017-07-09:
I suspect the `#include <Arduino.h>` at the beginning of the code that is messing the setup of NodeMCU somewhat

electrokean, 2017-07-09:
Are you removing the problematic yield() definition from the correct version of your HX711.cpp file
i.e. make sure you edit the copy in your Arduino libraries folder, and restart Arduino environment.
That seems to have fixed the issue for others: e.g. https://github.com/bogde/HX711/issues/69

zinkkrysty, 2017-07-09:
I am sure I am editing the right file because I tried to remove the `#include <Arduino.h>` and the program broke. 

I am trying to introduce the methods in the library one by one and see which one breaks the code.

So far only having in the headers doesn't break my program (also commented out the corresponding methods in the cpp file):

```
#ifndef HX711_h
#define HX711_h

#if ARDUINO >= 100
#include "Arduino.h"
#else
#include "WProgram.h"
#endif

class HX711
{
	private:
		byte PD_SCK;	// Power Down and Serial Clock Input Pin
		byte DOUT;		// Serial Data Output Pin
		byte GAIN;		// amplification factor
		long OFFSET = 0;	// used for tare weight
		float SCALE = 1;	// used to return weight in grams, kg, ounces, whatever

	public:
		HX711(byte dout, byte pd_sck, byte gain = 128);

		void begin(byte dout, byte pd_sck, byte gain = 128);
}
```

zinkkrysty, 2017-07-09:
Ok, I managed to fix this. 

The issue was that I reversed the pins when instantiating the scale (maybe [this tutorial](http://www.instructables.com/id/How-to-Interface-HX711-Balance-Module-With-Load-Ce/) was using older version of HX711 or they got it wrong)

Trying to fix it by commenting out the 

```
void yield(void) {};
```

actually causes the error for me when the pins are in correct order.

So the version that works for me:

```
HX711 scale(D6, D5); // D6 is DT, D5 is SCK
```

and with the library as is in the master of this repository (that has `void yield(void) {};` enabled)

Curiously enough, the difference between [my version](https://gist.github.com/zinkkrysty/3e0afdfb7c9a3cb219096555a4863d99) and the master is that the sides of the load cell are opposite (I have to use the other side to generate readings)

sravyachatti, 2018-02-26:
hi AgordiGuerrero sir,we are facing the same problem with hx711 code which you have faced 

(ets Jan 8 2013,rst cause:2, boot mode:(3,6)
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v60000318
~ld)

can you please explain in detail 
Thank you
waiting for your reply

Delirioo, 2018-04-15:
Hi
I'm trying to use a HX711 module on ESP8266 board following this example:
https://github.com/bogde/HX711/tree/master/examples/HX711Serial

but the serial monitor keeps getting this message:

`HX711 Demo
Before setting up the scale:
read: 		-411
read average: 		-289
get value: 		-116.00
get units: 		
Soft WDT reset

ctx: cont 
sp: 3ffef230 end: 3ffef490 offset: 01b0

>>>stack>>>
3ffef3e0:  3ffee438 00000000 3ffee438 00000000  
3ffef3f0:  00000000 ffffffb2 3ffee358 40201eb1  
3ffef400:  0000003c 0000000c 3ffe8490 402029c8  
3ffef410:  3ffe8460 00000007 3ffee438 0000000a  
3ffef420:  3ffee358 ffffffee 00000004 40202003  
3ffef430:  3ffe8484 3ffee358 3ffee438 3ffe8454  
3ffef440:  3ffe8460 3ffee358 3ffee358 402020b2  
3ffef450:  3ffe8460 3ffee358 3ffee438 40201d24  
3ffef460:  feefeffe feefeffe feefeffe 3ffee45c  
3ffef470:  3fffdad0 00000000 3ffee454 402027ec  
3ffef480:  feefeffe feefeffe 3ffee470 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)`

I can't understand were the mistake is. Thanks!

amotl, 2019-02-19:
Hi there, just wanted to let you know we are working on #123, which you might want to give a try.