bogde/HX711 #69: ESP8266 HTTP OTA failed when HX711.h is included

Werevolk, 2017-04-24:
Hi! First off all, many thanks for great and easy to use library!
Recently I came over an issue when trying to implement OTA update in my sketch. Everything works fine from the example sketch that I found here: https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/examples/WebUpdate/WebUpdate.ino
But as soon as I include <HX711.h> in the examle it crashes with that in serial:

> Soft WDT reset
> 
> ctx: cont 
> sp: 3fff03a0 end: 3fff0750 offset: 01b0
> 
> >>>stack>>>
> 3fff0550:  0000000a 3ffeec68 3fff0590 40203031  
> 3fff0560:  3ffe909c 3ffeec68 3ffeec68 4020765b  
> 3fff0570:  3ffe909c 3ffeec68 3fff0590 402076e3  
> 3fff0580:  00000000 3ffeec68 3ffeec44 40205122  
> 3fff0590:  3fff223c 0000000f 00000000 00000259  
> 3fff05a0:  3ffef63c 00000299 00000299 4010020c  
> 3fff05b0:  00000000 3ffe9188 00000001 3ffef63c  
> 3fff05c0:  00000030 3fff0650 00000000 4010053d  
> 3fff05d0:  40226c42 00000000 3fff243c 0000000f  
> 3fff05e0:  00000000 3fff0650 3fff0650 3ffe909c  
> 3fff05f0:  00000030 3fff0650 3fff0650 40207c7c  
> 3fff0600:  3fff26a4 00000026 3fff0650 40207ccb  
> 3fff0610:  3ffef63c 00000299 3fff0650 40207cfd  
> 3fff0620:  00000001 3ffeec68 3fff0650 40207e04  
> 3fff0630:  00000000 3ffeec68 3ffeec44 3ffe909c  
> 3fff0640:  00000001 3ffeec68 3ffeec44 402055e2  
> 3fff0650:  3fff2574 0000002f 00000026 3fff275c  
> 3fff0660:  0000008f 00000017 3fff25e4 0000001f  
> 3fff0670:  0000000f 3fff26a4 0000002f 00000026  
> 3fff0680:  3fff2294 0000000f 00000000 3fff227c  
> 3fff0690:  0000000f 00000000 3fff2264 0000000f  
> 3fff06a0:  00000007 3fff2534 0000000f 00000004  
> 3fff06b0:  3fff260c 0000008f 00000000 40202ef8  
> 3fff06c0:  00000001 3ffef514 0003c876 00000000  
> 3fff06d0:  00000000 40203518 0000000f 40202e82  
> 3fff06e0:  00000000 00000000 3ffeec44 3ffef720  
> 3fff06f0:  00000257 3ffeec68 3ffeec44 4020443f  
> 3fff0700:  3ffe8550 00000000 000003e8 3ffeec44  
> 3fff0710:  00000000 3fff254c 3ffef664 40202282  
> 3fff0720:  3fffdad0 00000000 3ffef718 402022a0  
> 3fff0730:  3fffdad0 00000000 3ffef718 40208420  
> 3fff0740:  feefeffe feefeffe 3ffef730 40100718  
> <<<stack<<<
> 
>  ets Jan  8 2013,rst cause:2, boot mode:(3,7)
> 
> load 0x4010f000, len 1384, room 16 
> tail 8
> chksum 0x2d
> csum 0x2d
> v09f0c112
> ~ld
> ⸮
> 

I'm using NodeMCU v3 by Lolin for tests.
That happens even when there is no scale.begin in code, only library include. For now I disabled OTA functionality, but hopefully this issue could be fixed.

Sorry for my bad English :)

bogde, 2017-04-24:
can you share your code?

Werevolk, 2017-04-24:
It's a standard example:
`
#include <HX711.h>

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <ESP8266HTTPUpdateServer.h>
#include <DHT.h> //  DHT.h library https://github.com/markruys/arduino-DHT

const char* host = "esp8266-webupdate";
const char* ssid = "SSID";
const char* password = "PASS";

ESP8266WebServer httpServer(80);
ESP8266HTTPUpdateServer httpUpdater;

void setup(void){

  Serial.begin(115200);
  Serial.println();
  Serial.println("Booting Sketch...");
  WiFi.mode(WIFI_AP_STA);
  WiFi.begin(ssid, password);

  while(WiFi.waitForConnectResult() != WL_CONNECTED){
    WiFi.begin(ssid, password);
    Serial.println("WiFi failed, retrying.");
  }

  MDNS.begin(host);

  httpUpdater.setup(&httpServer);
  httpServer.begin();

  MDNS.addService("http", "tcp", 80);
  Serial.printf("HTTPUpdateServer ready! Open http://%s.local/update in your browser\n", host);
}

void loop(void){
  httpServer.handleClient();
}
`

It works fine if I'm not including HX711 library but I get WDT reset if I'm trying to perform OTA with included library.

electrokean, 2017-04-24:
Sounds similar to #63 - not sure what, but something in the include apparently upsets the ESP network libraries.
@Werevolk Can you try moving the include of HX711.h after all the other includes?

Werevolk, 2017-04-24:
Just tried to move include after all the others. Not helped at all, same error on OTA:

> Soft WDT reset
> 
> ctx: cont 
> sp: 3fff0380 end: 3fff0750 offset: 01b0

Werevolk, 2017-04-24:
I think I've found the solution!
I cpmmented out yield() in H711.cpp
`
#if ARDUINO_VERSION <= 106
    // "yield" is not implemented as noop in older Arduino Core releases, so let's define it.
    // See also: https://stackoverflow.com/questions/34497758/what-is-the-secret-of-the-arduino-yieldfunction/34498165#34498165
   // void yield(void) {};
#endif
`

And finally OTA is able to finish without WDT resets!

As I understand it also will fix an issue causing resets when module is failed to initialize at setup().

kdeklvadiya, 2017-07-17:
@werevolk
i will be very great full to you if you please upload the code which have hx711 & dth sensor for nodemcu.

cnuma, 2020-04-22:
Hello 

Thanks for this publish.
To stop hangUp of my ESP8266, i've little modified HX711.cpp and HX711.h

On HX711.h, i've add new variable named : unsigned long YIELDMILLIS;
And, on HX711.cpp, i've:

1. Comment on top file line 7 with //void yield(void) {};
2. On function <<long HX711::read() {>>, i've little modified starting of it by:
long HX711::read() {
	
	// wait for the chip to become ready
	YIELDMILLIS = millis();
	while (!is_ready()) {
		// Will do nothing on Arduino but prevent resets of ESP8266 (Watchdog Issue)
		yield();
		if (millis() - YIELDMILLIS > 500) { //Try to fetch response for 5
		  return -1;
		}
		
	}

This new, don't hangup my loop ;-)

Bye !
Manu