bogde/HX711 #71: Tare

flc85, 2017-06-27:
It's possible to use more than 1 tare?

electrokean, 2017-06-27:
Not from the library.  But the tare is just a zero offset.
You can save your own offsets in your code and do the simple math (subtraction).

flc85, 2017-06-27:
for more explication:

I put a container -> tare1
I put products in the container and it becomes too small.
I put another container over -> tare2

I'm a newbie with arduino some explications how to save the offsets.

electrokean, 2017-06-27:
Sorry, but this isn't the place to teach general programming.
The library gives you weights in units - the tare "offset" is just a nominal weight that is a zero reference, which you can subtract from later weight readings to get the difference.

flc85, 2017-06-27:
I understand thank you for your help

flc85, 2017-07-01:
in the example:
 scale.set_scale(21680.f);
why .f ?

electrokean, 2017-07-01:
It forces the constant to be floating point.  Not critical in this case, but good practice.

electrokean, 2017-07-01:
For a little more info, see https://stackoverflow.com/questions/5026570/suffix-of-f-on-float-value

flc85, 2017-07-02:
ok thanks for the information.
It's possible to limite the result of scale.get_units(10); at 2 decimal?

flc85, 2017-07-03:
I put a container -> tare1 
scale.tare()
i read get_offset() and i put it in a variable T1
I put products in the container and it becomes too small.
I put another container over -> tare2
scale.tare()
i read get_offset() and i put it in a variable T2
i read the scale get.value and to know the weight of products:   ((get.scale value-(T1-T2))/set_scale)
It is correct?

electrokean, 2017-07-03:
No, that doesn't make sense to me.

You use `scale.tare()` only once, and then used `scale.get_units()` to get the weight since tare.
I'm assuming you have called `scale.set_scale()` properly.

Proper way:
1. put on container
2. call `scale.tare()`
3. `scale.get_units()` gives you W0 - should be 0 (or very close)
4. put product in container
5. `scale.get_units()` gives you W1 - contents so far
6. put another container
7. `scale.get_units()` gives you W2 - and (W2 - W1) is weight of second container
8. add more product
9. `scale.get_units()` gives you W3 - which is total weight since tare
10. W3 - (W2 - W1) is total product weight

If you didn't do the tare then W0 wouldn't be 0, but you could just subtract W0 from total.

flc85, 2017-07-03:
Good idea, thank you

closed 2017-12-06 by bogde