bogde/HX711 #72: feather 32u4 problem

paolometeo, 2017-07-08:
Hi, 
I found trouble with a HX711 connected to a Feather 32u4 (I tested various pin). It seems to not communicate.
The same circuits work well with an Arduino board, as 5 V or 3.3 V. Did anyone have this problem?
Paolo

ashotaro, 2019-01-16:
My PICO ( ATmega32u4 ) can communicate with my laptop, but the sensor value look incorrect.  
I uploaded same sketch to it and my UNO, but only UNO works correctly.  
My sketch is here:  
 `#include "HX711.h"

#define DOUT  3
#define CLK  2

HX711 scale(DOUT, CLK);

float calibration_factor = -7050; //-7050 worked for my 440lb max scale setup

void setup() {
  Serial.begin(9600);
  Serial.println("HX711 calibration sketch");
  Serial.println("Remove all weight from scale");
  Serial.println("After readings begin, place known weight on scale");
  Serial.println("Press + or a to increase calibration factor");
  Serial.println("Press - or z to decrease calibration factor");

  scale.set_scale();
  scale.tare(); //Reset the scale to 0

  long zero_factor = scale.read_average(); //Get a baseline reading
  Serial.print("Zero factor: "); //This can be used to remove the need to tare the scale. Useful in permanent scale projects.
  Serial.println(zero_factor);
}

void loop() {

  scale.set_scale(calibration_factor); //Adjust to this calibration factor

  Serial.print("Reading: ");
  Serial.print(scale.get_units(), 1);
  Serial.print(" lbs"); //Change this to kg and re-adjust the calibration factor if you follow SI units like a sane person
  Serial.print(" calibration_factor: ");
  Serial.print(calibration_factor);
  Serial.println();

  if(Serial.available())
  {
    char temp = Serial.read();
    if(temp == '+' || temp == 'a')
      calibration_factor += 10;
    else if(temp == '-' || temp == 'z')
      calibration_factor -= 10;
  }
}
`
On ATmega32u4 board, the sensor value of load cell changes by about 32,2 lbs constantly.

Does anyone tell me how to communicate with ATmega32u4 correctly?

bogde, 2019-02-24:
i just tested the current code on Feather 32u4 and it seems to work well, you may want to give it another try.