bogde/HX711 #76: Sketch lockup when using constructor with the parameters and HX711 not connected

Petorrr, 2017-07-28:
When using the constructor to initialize the class with parameter passing, the system locks up when my sensor board is not connected. I found out that this is caused by the 'read' function that is called through the set_gain/begin calls. The while loop in that function stays there forever, thus hanging the system since the system is not yet through it's setup.

Just a proposed improvement, that will make the read more robust against 'hanging' in a loop;

byte retry = 0;

	// wait for the chip to become ready
	while (!is_ready() && retry < 20)
	{
		delay (10);
		retry++;
		// Will do nothing on Arduino but prevent resets of ESP8266 (Watchdog Issue)
		yield();
	}
    if (retry >= 20)
	  return (0);

lewsut, 2018-12-22:
This absolutely solves the stuck in the while loop I was having when the scales were disconnected or failed.

amotl, 2019-02-19:
Dear Peter,

while working on #125 in the aftermath of #123, we just added your wait-for-readyness-with-retry implementation to our [spring-cleaning branch](https://github.com/hiveeyes/HX711/tree/spring-cleaning) by 87727e0.

The corresponding [HX711_retry_example.ino](https://github.com/hiveeyes/HX711/blob/0.7.0/examples/HX711_retry_example/HX711_retry_example.ino) uses the new `wait_ready_retry()` function originally coming from your suggestion.

With kind regards,
Andreas.

gjt211, 2020-03-15:
I think I am having the same problem using the most recent version of code. If I power up the board without the HX711 connected it locks up.

```
scale.begin(LOADCELL_DOUT_PIN, LOADCELL_SCK_PIN);
```

I can't work out how to fix this issue, it would be good if the `scale.begin` could timeout and return a success or fail.