bogde/HX711 #78: Update license to LGPL to allow use in proprietary products

frankleonrose, 2017-08-23:
@bogde, would you consider updating the license to LGPL or MIT? The current GPL license explicitly prohibits use of the code in proprietary products. Wouldn't you rather your code were available to everyone to use?

bogde, 2017-08-29:
i'll think about it. anyone else has any thoughts on this?

rbicelli, 2018-07-17:
I'm developing a proprietary project that uses your library, for part of the system. I would like to maintain part of project proprietary and release some libraries under MIT license. Since your library is released under GPL, I will forced to release my entire project, not only parts of it, under GPL.

frankleonrose, 2018-07-23:
Polling contributors per @bogde 's concern: @zfields @SirUli @Gulaschcowboy @amotl @Deamiter

Will you allow updating the HX711 repo to a more permissive license as discussed? 
(No response will be assumed consent.)

SirUli, 2018-07-23:
Fine with me.

Cheers,
Uli

zfields, 2018-07-24:
I love it! Go OSS!!!

frankleonrose, 2018-11-28:
@bogde Thanks very much!

bogde, 2018-11-28:
@frankleonrose you're welcome

i'll also try if time permits to do some work on this lib, maybe pull some of the pull requests and such

ulno, 2018-11-28:
Also thanks from me, I couldn't make HX711_ADC work and this library does, so I will rewrite my framework now to include this here.