bogde/HX711 #82: calibrate my load cell by Placing a unknown weight (tare)

MisterObe, 2017-09-19:
Hello, 

I work on a project,  and I've set up my load cell and tried the calibration process.
I'm using arduino mega, hx711 and 4pcs of 5kg load cell.

For the first step, I already calibrated my load cell thanks to this tuto (whith a know weight... by Placing a known weight on the scale and call get_units(10) and by Dividing the result in step 3 to my known weight) : https://github.com/bogde/HX711

And for the next step I would like to calibrate my load cell by Placing a unknown weight on the scale from the beginning... create a loop that does it automatically.

Do you tkink it's possible to do that ? If it's possible, do you have any approach to do that ?

Thanks,

bogde, 2017-09-20:
what's your use case?

note that once you calibrate your scale either by using a known weight or by working out the math, you don't need to do it again. you just store the resulting value in your sketch and use that every time. so you don't need to run the calibration procedure every time, you can just comment it out. you can't calibrate using an unknown weight.

MisterObe, 2017-09-21:
Hello, thank you for your reply

 I use it for automatic dosing of paint.

I don't want to run the calibration procedure every time, just on the beginning  of my program.

Actually I calibrate my scale either by using a known weight, and I would like to do it by working out the math (weight unknow).

Here is my desired approach : 

- step one : Place a unknown weight on the scale

- step twoo : Calibrate my scale
... Suite of my program

Do you tkink it's possible to do that ?
Thanks,

bogde, 2017-09-21:
see this thread: https://github.com/bogde/HX711/issues/20

however, i still don't understand why you want to calibrate every time. calibration is only used in order to get the value that's passed to set_scale, like scale.set_scale(2280.f). once you get that, one way or another, you can simply pass the same value to set_scale every time your sketch starts.

in your step one above, do you want to actually measure the unknown weight right? in that case, you need to use scale.set_scale(....) before that, using the value that you already obtained using a known weight.

bogde, 2017-12-06:
since there wasn't any feedback on this, i'm closing the issue. feel free to open it again if needed.

closed 2017-12-06 by bogde

sgvs-deloitte, 2023-08-09:
Hello, the situation is the same for me.
I observe an electric drift or mechanical disturbances creating weight calculation inaccuracies when the load cell is continuously used. To fix it, one should recalibrate with a known weight before continuing. I don't want my consumers to experience it. I want my firmware to detect the need for calibration and perform it with an unidentified weight. Then It consistently produces accurate data.