bogde/HX711 #83: Sometimes "wrong" (remarkable) values are measured

eTh0maz, 2017-09-24:
I'm logging weight every 15 minutes, part of a permanent scale project.
In these loggings sometimes remarkable "wrong" values are returned (using scale.read() or scale.read_average(x)), resulting in a spike in the graph's.

8388607 (= 2^23 - 1)
4194303 (= 2^22 - 1)
2097151 (= 2^21 - 1)
1048575 (= 2^20 - 1)
...
Sometimes it's also just "0"

It's always a value that can be calculated by 2^X - 1. Always the double value than the previous one.
I guess this might be a library related issue? I tried multiple HX711 modules, load cells and measurement approaches in software but the issues stayed the same.
The hardware is Arduino pro mini based, 8Mhz operating on 3.3V with a HX711 module & one loadcell.

electrokean, 2017-09-25:
Not sure what is going on for you, but I suggest you try doing a dummy read, throw away the value, and then do your actual read.  See if that filters out the wrong values.

eTh0maz, 2017-09-25:
These errors occur in only 1 to 5 percent of the logged measurements.
At the moment I'm taking a single measurement and a second later a second (average) measurement to check if the load cell is stable before continuing.
This means both first and second average measurements are the same (+- a 250 threshold).

I could filter out these "wrong" value's and measure again but this doesn't solve the initial problem. Because it's a battery powered system designed to operate for as long as possible I can't afford to repeat the measurement.

electrokean, 2017-10-08:
What are you doing to save power?  Do you power down the HX711?  It sounds like a timing issue.

bogde, 2017-12-06:
since there wasn't any feedback on this, i'm closing the issue. feel free to open it again if needed.

closed 2017-12-06 by bogde

Fusseldieb, 2021-05-19:
This issue still persists, and not only with this library. I think that's a fault in the chip design itself.

```
170.75
170.83
171.22
170.80
170.95
171.04
-143.37
171.36
170.72
170.86
171.10
170.81
170.90
171.05
170.81
```

ianmackinnon, 2025-08-22:
I was seeing precisely these `2^X - 1` values and the issue was a faulty E+ connection.