bogde/HX711 #88: No auto tare on start

maxholgasson, 2017-12-05:
Hello,  after searching for hours..  
I need the HX711 for a beehive scale. Using the esp8266 which is in deep sleep and wake up every 10 minutes (solar powered, so low energy needed).

How can I remove this line of code
scale.tare(); //Reset the scale to 0
and replace it so that the scale continues reading the weight instead of setting it to 0 on start.

bogde, 2017-12-06:
can't you just remove that line?

Pim-Veld, 2017-12-25:
I think you will find your answer here:

https://github.com/sparkfun/HX711-Load-Cell-Amplifier/blob/master/firmware/SparkFun_HX711_KnownZeroStartup/SparkFun_HX711_KnownZeroStartup.ino
Further this page may be of interest to you (perhaps you have already seen it):

https://makezine.com/projects/bees-sensors-monitor-hive-health/

closed 2019-02-21 by bogde