bogde/HX711 #91: Anyone tried to use a HX712

tk5ep, 2017-12-20:
Hi all,
I finally found the time to build a board with an Arduino pro mini and four HX712. 
Unfortunately, i can't get anything from these chips.. I used the HX711 library as it should work, at least with gain 128. (the HX712 should be able to have a gain of 256)
I can't get anything stable and no variation at all when putting some weight on the cell. If i unbalance the differential inputs with a resistor, i can get something. But that would correspond to a VERY heavy weight. It is like if the gain is VERY limited.
Has anyone tested these chips or are they just crap or fake ?

tk5ep, 2017-12-22:
I answer myself... It was a hardware problem. The AVDD supply was not the same as the REFP ! It works fine now at gain 128.
But, i still can't switch the chip to gain 256... Seems to be an internal problem.

tk5ep, 2017-12-25:
Hi all,
Just for your information. I wrote to Avia company asking if the batch of HX712 i have are really coming from their plant and if the HX712 has a gain of 256.
Here their answer ! 
It's really disappointing to read that a manufacturer is advertising for a chip and does not follow the specifications !
So, stay away from these HX712. There is no reason to use it ! It has worse than the HX711 in all respects... I learned it the hard way.
The good news, is that th HX711 library is working 100% with the HX712.

-----------------------------------------------------------------
Hi,
This batch is coming from our factory. HX712 can' t be used at a gain of 256.
Thank you.
Best regards,
Mandy
海芯科技（厦门）有限公司
AVIA SEMICONDUCTOR (XIAMEN) CO. , LTD.
ADD: 301E NO. 34 Guanri RD, Xiamen Software Park Ⅱ, Xiamen, China 
Tel: 0592-2529530
Fax: 0592-5996329

Darktidelulz, 2018-01-05:
tk5ep thanks for the update!

closed 2018-11-28 by bogde