bogde/HX711 #92: Continuous measurement slowly increases

daleyjem, 2018-01-04:
Basically, if I calibrate for 10 pounds of weight... After the first `.tare()` is called on `setup()`, the numbers are correct, but if I let that sit for about 45 minutes, there appears an increase of about 3 pounds.

I imagine this has something to do with the load cells themselves, but is there any way you can think of to offset these increases over time?

tk5ep, 2018-01-04:
Well, 3 pounds increase for a 10 pounds initial weight is A LOT !! 30 % drift is clearly abnormal.
Before trying to correct this by software, you should check your hardware !

daleyjem, 2018-01-04:
It's basically a repurposed bathroom scale. Those things typically timeout after less than 30 seconds, and tare every time you stand on them again. So after 45 minutes, is it actually a lot? Can you say with certainty that 50kg load cells maintain accuracy after continuous weight?

Gambit120, 2018-01-04:
I have been experienceing the same thing. I am trying to measure small changes over a long time. Like the weight of a wet sponge as the water evaporates. After a day my scales drift 5-10 grams.

tk5ep, 2018-01-04:
You probably use [cheap weight cells like these](https://www.sparkfun.com/products/10245) 
They are not made for permanent load, but in no case should they drift that much ! Many people use them with permanent load with fair results.
I use other (better) cells and during my tests with 40 kgs, they were stable for weeks within a few grams ...

Darktidelulz, 2018-01-05:
I've got even cheaper cells (4 cells $4,-), there has been a 20kg calibration weight on there for the past half year and it drifts with temperature (-250 to +250grams).

Do you have your wires soldered onto the HX711? (This is a must!)
How did you do the wiring of your wheatstone bridge? (Had problems with wrong wiring)
What Kind of power source are you using? (experienced problems when it was on a USB-hub connected to a USB-hub)

Could you post your setup() and the HX711 part of the loop?

daleyjem, 2018-01-05:
Sure...

```c++
/**
 * This file is the source for the hardware if using the scale for measurement.
 */

// Scale library
#include "HX711.h";
// Bluetooth library
#include <SoftwareSerial.h>;

#define ACKNOWLEDGE_TIMER 10000
#define BT_TYPE_WEIGHT 1
// This is an 'awake' signal to let the device know it can still communicate over bluetooth
#define BT_TYPE_ACKNOWLEDGE 3
#define BT_RX_PIN 4
#define BT_TX_PIN 2
#define SCALE_DT_PIN 9
#define SCALE_SCK_PIN 10

signed int acknowledgeCounter = 0;
float calibration_factor = -10950.00;

HX711 scale( SCALE_DT_PIN, SCALE_SCK_PIN );
SoftwareSerial mySerial( BT_RX_PIN, BT_TX_PIN );

void setup() {
  scale.set_scale( calibration_factor );
  scale.tare(); //Reset the scale to 0
  
  Serial.begin( 38400 );

  // The HC-06 defaults to 9600 according to the datasheet.
  mySerial.begin( 9600 );

  useInterrupt( true );
}

void loop() {
  bt_weight( scale.get_units() );
}

// Interrupt is called once a millisecond.
// This is like calling window.setInterval() in Javascript.
SIGNAL(TIMER0_COMPA_vect) {
  acknowledgeCounter++;
  
  if (acknowledgeCounter == ACKNOWLEDGE_TIMER) {
    bt_acknowledge();
    acknowledgeCounter = 0;
  }
}

/**
 * Sends a bluetooth scale reading
 */
void bt_weight(float reading) {
  Serial.println((String)reading);
  mySerial.print((String)BT_TYPE_WEIGHT + ":" + (String)reading + "\n");
}

/**
 * Sends a bluetooth acknowledgement pulse to let device know BT connection is still alive
 */
void bt_acknowledge() {
  mySerial.print((String)BT_TYPE_ACKNOWLEDGE + "\n");
}

void useInterrupt(boolean v) {
  if (v) {
    // Timer0 is already used for millis() - we'll just interrupt somewhere
    // in the middle and call the "Compare A" function above
    OCR0A = 0xAF;
    TIMSK0 |= _BV(OCIE0A);
  } else {
    // do not call the interrupt function COMPA anymore
    TIMSK0 &= ~_BV(OCIE0A);
  }
}
```

MBENKI, 2021-04-14:
> You probably use [cheap weight cells like these](https://www.sparkfun.com/products/10245)
> They are not made for permanent load, but in no case should they drift that much ! Many people use them with [...]

I have the same problem, but the drift is only 30 grams (in my project it is already too much) and i use those cheap cells, could you please suggest those better cells you use? Thank you so much