bogde/HX711 #93: No readings from HX711 Configuration.

armaldro, 2018-01-08:
Dear Github Members,

I am new to using Arduino and have reach an issue that I couldn't resolve despite looking through the forum and attempting solutions for similar issues that others have succeeded .

I'm currently attempting to get a reading from HX711 amplifier connected to a TAL220 (10KG bar type load cell) using Arduino.

[image]

```
#include "HX711.h"

// HX711.DOUT	- pin #A1
// HX711.PD_SCK	- pin #A0

HX711 scale(A1, A0);		// parameter "gain" is ommited; the default value 128 is used by the library

void setup() {
  Serial.begin(38400);
  Serial.println("HX711 Demo");

  Serial.println("Before setting up the scale:");
  Serial.print("read: \t\t");
  Serial.println(scale.read());			// print a raw reading from the ADC

  Serial.print("read average: \t\t");
  Serial.println(scale.read_average(20));  	// print the average of 20 readings from the ADC

  Serial.print("get value: \t\t");
  Serial.println(scale.get_value(5));		// print the average of 5 readings from the ADC minus the tare weight (not set yet)

  Serial.print("get units: \t\t");
  Serial.println(scale.get_units(5), 1);	// print the average of 5 readings from the ADC minus tare weight (not set) divided 
						// by the SCALE parameter (not set yet)  

  scale.set_scale(2280.f);                      // this value is obtained by calibrating the scale with known weights; see the README for details
  scale.tare();				        // reset the scale to 0

  Serial.println("After setting up the scale:");

  Serial.print("read: \t\t");
  Serial.println(scale.read());                 // print a raw reading from the ADC

  Serial.print("read average: \t\t");
  Serial.println(scale.read_average(20));       // print the average of 20 readings from the ADC

  Serial.print("get value: \t\t");
  Serial.println(scale.get_value(5));		// print the average of 5 readings from the ADC minus the tare weight, set with tare()

  Serial.print("get units: \t\t");
  Serial.println(scale.get_units(5), 1);        // print the average of 5 readings from the ADC minus tare weight, divided 
						// by the SCALE parameter set with set_scale

  Serial.println("Readings:");
}

void loop() {
  Serial.print("one reading:\t");
  Serial.print(scale.get_units(), 1);
  Serial.print("\t| average:\t");
  Serial.println(scale.get_units(10), 1);

  scale.power_down();			        // put the ADC in sleep mode
  delay(5000);
  scale.power_up();
}
```
Would be great if I could get some advice as to why I'm not getting any results and do let me know if I can provide more information to get assistance. Thank you for taking your time to assist me in my journey.

tk5ep, 2018-01-08:
try reversing  pin A0 and A1...

armaldro, 2018-01-09:
Hi tk5ep,

thank you for the advice.
tried reversing, still doesn't work. on my side, I've thought that it was the connection from the load cell to the hx711 being loose so i've bought the terminal and ratchet them on.

tk5ep, 2018-01-09:
Even without any cell connected, you should see some datas coming out,
probably close to 0.00

Being a newcomer, are you sure your sketch is correctly loaded in the
Arduino ? have you tried other sketches ? What do you see on your terminal
screen ?

On your picture, there is a wire going to Vcc of the HX711 board. Is it
connected to the 5V or 3.3V source of the Arduino ?

armaldro, 2018-01-09:
Hi tk5ep,

There are readings coming out, but all are 0.00
I've connected it to the 5V source.

Will update the next reply with the terminal screen

armaldro, 2018-01-09:
[image]

tk5ep, 2018-01-09:
Well, the only wrong thing i saw is that you physically inverted the DOUT and PD_SCK pins. DOUT is declared on pin A1 and according your diagram, you wired it on A0.
But i already asked you to correct this...
If you have a voltmeter, you could measure the voltages on the HX711 board. First between GND and VCC + VDD ( 5V ) and between RED and BLACK.
The second one MUST switch from 0V to about 4V depending of power_down and power_up. If this happens, you're sure that at least the Arduino communicates with the HX711.
Between GREEN and WHITE, you should have half of the 4V voltage.

Have you another HX711 board ?

chomamichalzmf13, 2018-01-15:
try remove tare function, or check above my issue. What color are wires on your load cell? I had one, with blue, black, red, white, and pinout is different. Black connect to A+, blue to A-, white to E-, red to E+.

armaldro, 2018-01-24:
Thank you @tk5ep and everyone!
So the issue with my board was that my Arduino Board jumper wire connecting the VCC and VDD for HX711 was not constant at 5V causing it to be 0.
This was fixed after changing 2 arduino board.

Thank you very much for giving me assistance and guidance over my project :)

tk5ep, 2018-01-24:
Great if you solved your problem.
Good luck in your project

Sevda-Aliparast, 2018-05-08:
Hi everyone
I have the same problem with my loadcell project, I've comments through solve it. But it still shows me 0 output.

armaldro, 2018-05-09:
Do you have your connection diagram?
Some of the troubleshooting guides I follower.

-> Ensure all pins are properly soldered and connected.
-> Ensure that VCC is supplied with the appropriate voltage. (5V in my project which I set up with a pull-up resistor.
-> Ensure that the load cell is mounted properly.

Sevda-Aliparast, 2018-05-09:
I did it thanks, but now it shows meaningless numbers on screen.
What can be problem?
I copied ur code to my arduino

armaldro, 2018-05-09:
If it is displaying weird characters.
You may need to change the baud rate displayed on your serial monitor.
Alternatively, you can try to use this code.

`#include "HX711.h"

// HX711.DOUT	- pin #A1
// HX711.PD_SCK	- pin #A0

HX711 scale(A1, A0);		// parameter "gain" is ommited; the default value 128 is used by the library

void setup() {
  Serial.begin(9600);
  Serial.println("HX711 Demo");

  Serial.println("Before setting up the scale:");
  Serial.print("read: \t\t");
  Serial.println(scale.read());			// print a raw reading from the ADC

  Serial.print("read average: \t\t");
  Serial.println(scale.read_average(20));  	// print the average of 20 readings from the ADC

  Serial.print("get value: \t\t");
  Serial.println(scale.get_value(5));		// print the average of 5 readings from the ADC minus the tare weight (not set yet)

  Serial.print("get units: \t\t");
  Serial.println(scale.get_units(5), 1);	// print the average of 5 readings from the ADC minus tare weight (not set) divided 
						// by the SCALE parameter (not set yet)  

  scale.set_scale(2280.f);                      // this value is obtained by calibrating the scale with known weights; see the README for details
  scale.tare();				        // reset the scale to 0

  Serial.println("After setting up the scale:");

  Serial.print("read: \t\t");
  Serial.println(scale.read());                 // print a raw reading from the ADC

  Serial.print("read average: \t\t");
  Serial.println(scale.read_average(20));       // print the average of 20 readings from the ADC

  Serial.print("get value: \t\t");
  Serial.println(scale.get_value(5));		// print the average of 5 readings from the ADC minus the tare weight, set with tare()

  Serial.print("get units: \t\t");
  Serial.println(scale.get_units(5), 1);        // print the average of 5 readings from the ADC minus tare weight, divided 
						// by the SCALE parameter set with set_scale

  Serial.println("Readings:");
}

void loop() {
  Serial.print("one reading:\t");
  Serial.print(scale.get_units(), 1);
  Serial.print("\t| average:\t");
  Serial.println(scale.get_units(10), 1);

  scale.power_down();			        // put the ADC in sleep mode
  delay(5000);
  scale.power_up();
}`

Sevda-Aliparast, 2018-05-09:
thank you very much
I think I did it, :)  surely with your valuable comments...
I should connect the loadcell, arduino to wifi module ( ESP 8862), Have you
had any experience with the case?

closed 2019-02-22 by bogde

fenixil, 2024-02-23:
In my case I had to introduce a delay after scale is powered up, otherwise scale was not ready and returned zeroes:
```
scale.power_down(); // put the ADC in sleep mode
delay(5000);
scale.power_up();
scale.wait_ready_timeout(500)
// ^^^^ here
```
https://github.com/bogde/HX711/issues/251#issuecomment-1960681540