bogde/HX711 #94: tare function

chomamichalzmf13, 2018-01-15:
Hello, it is possible to fix "tare" function? Or try to check it. I used your example for calibrate load cell 3 kg with HX 711 module (serial example) and values osscilates, not work, module dont response on weight. I remove tare function and it works. 
Also possible problem can be with temperature balancer which is add to load cell (check photo, verticall label), but dont know. If someone has problem ,try to remove tare function and measure resistance from red wire to others wires. I short circuit (see photo on temperature balancer).
I used Arduino Uno and buy load cell with red, black, white and blue wire, also wiring is little different. Red wire is E+, 
Black wire is A+
White wire is E-,
Blue wire is A-

A- and A+ can be reverse, if output is negative
E- and E+ can be reverse if output is negative.
So problem is only, when power and signal wire are mixed together.

[image]
[image]
[image]
[image]
[image]
[image]

Pim-Veld, 2018-02-17:
First of all, go the the "Code" tab and read the content of file HX711.h. It contains a detailed explanation of every function of the library.

When you refer to vertical "label" I assume you mean the vertical "strain gauge" in your pictures. Both strain gauges should be incorporated in a bridge of Wheatstone to even out changes in resistance due to temperature differences. Which is precisely what circuit board HX711 is supposed to do (amongst others), provided the wires of the strain gauges are properly connected.

Nevertheless there are other reasons why the output of the HX711 board can fluctuate. One of them is induction of noise picked up by the strain gauges and the connection wires and probably also noise generated in the circuits on the HX711 board. Therefore it is advisable to take the average of several readings. For several functions in the library this is very easy to do by setting the parameter (between the parenthesis) to the number of readings to average over.

The call to scale.begin(3, 2) is unnecessary as you have already set this in the constructor:
HX711 scale(DOUT, CLK); (with DOUT=3 and CLOCK=2).

The function tare() sets the OFFSET parameter to the currently measured weight. This causes function get_value() and consequently also function get_units() to return zero (see HX711.h in de code section). Therefore you should include this function in the setup section of your sketch and not in the loop section. It is advisable to average over a number of readings, for example 20: tare(20). I am not sure why, but in seems common practice to include a considerable delay (f.e. 1 second) after the call to the function tare(20).

The function set_scale() sets the SCALE parameter to the float value between the parenthesis. Omitting a value sets SCALE = 1.0 You should set SCALE to such a value that the output from function get_units() is what you want it to be. For example: you have 1 kg on the scale and want an output in grams, then you have to set SCALE to such a value that get_units() returns 1000. Another example: you have 5 kg on the scale and want an output in kg then you have to set SCALE to such a value that get_units() returns 5
You should include this function in the setup section of your sketch and not in the loop section. Also here it is advisable to include a delay (experiment with it ).

Finally, in de loop section, the function get_units() should give you the "clean" weight, without the need to do any further corrections. And, as you did, it is advisable to average over a number of readings: get_units(10);

bogde, 2018-11-28:
@chomamichalzmf13 i'm going to close this issue shortly, but if you didn't solve your issue feel free to open it again.

closed 2018-11-28 by bogde