bogde/HX711 #99: Can Hx711 module be used with other sensors?

SreevatsaAcharya, 2018-03-07:
Hi,
I am using a deflection based axle load sensor shown below

http://sensomaticloadcell.net/deflection-sensor/

Can i use the Hx711 module and this library to interface with a raspberry pi?

SreevatsaAcharya, 2018-03-14:
anyone?? @bogde ?

bogde, 2018-03-14:
the library was written for arduino and compatible devices. i assume you could port it to run on the raspberry pi, if thats what you're asking.

SreevatsaAcharya, 2018-03-14:
what about the sensor? It is a strain gauge based sensor but not a load cell. It gives out an output based on slightest deflections. I have managed to get values setting up with arduino, but have no idea how to calibrate it. Also planning to add 4 sensors in parallel and sum the values.Any advice is appreciated.

lucmestchen, 2018-03-17:
If the sensor is strain gauge based you have to connect it in half bridge or full bridge.
You can use: 

- one sensor and one fixed resistor (half bridge). 
- two sensors  (half bridge).   
- four sensors (Wheatsone bridge)

Maybe links below are a help for you? 

https://www.transducertechniques.com/wheatstone-bridge.aspx

https://learn.sparkfun.com/tutorials/load-cell-amplifier-hx711-breakout-hookup-guide?_ga=2.39941803.809879608.1521259190-441915726.1519060593

SreevatsaAcharya, 2018-03-21:
hey @lucmestchen ,@bogde,
I am planning to use a half bridge configuration for my setup but am confused about the connections.
The sensor uses 10 - 15 v input excitation voltage and this the red wire is already occupied.
what connections can i give to E+ and A+?  Should i use only white wires and give to A+ and leave E+ unconnected? Would you recommend this setup for accuracy ?

lucmestchen, 2018-03-21:
Do you have a datasheet of the sensor?
The pdf link on this address  from you:  http://sensomaticloadcell.net/deflection-sensor/    is not working.

If you use a half bridge loadcell (2 resistors) the output  in rest (no load) will be  1/2 of your exitation voltage (5V when you use 10V).

The HX711 is build for very small voltages.  You can make an additional voltage divider with two fixed resistors.  Then the voltage difference between the center of the two divders is 0V in rest.
If you bent the loadcell, the voltage of this dividers will change a little bit.  There is a small difference in voltage between te two dividers.  This is applied to the amplifier in the HX711.

Please read this if you don't understand voltage dividers.  A bridge is nothing more than two dividers.
https://learn.sparkfun.com/tutorials/voltage-dividers

SreevatsaAcharya, 2018-03-21:
[BS.pdf](https://github.com/bogde/HX711/files/1833251/BS.pdf)

lucmestchen, 2018-03-21:
Red is your exitation voltage +10V or +15V
Black is 0V or GND of the exitation voltage.
You can connect shield together with black.  
Green and white to the amp input + and -. If you reverse green and white sign of youre output reading would normally change.

closed 2018-04-19 by bogde