bogde/HX711 #101: soft WDT reset on Esp8266

cbwarren, 2018-04-12:
I've been troubleshooting soft resets on project using scale and esp8266; when I comment out the HX711 library and related commands and substitute fixed values for the reading, the soft reset issue goes away!

cbwarren, 2018-04-12:
further testing: if all i do is add " #include HX711", the i get the following : 
Soft WDT reset

ctx: cont 
sp: 3fff0780 end: 3fff0ab0 offset: 01b0

>>>stack>>>
3fff0930:  0000005b 00000000 3fff2724 402172fb  
3fff0940:  d8531795 00000000 00000000 00000026  
3fff0950:  0000005b 3fff2949 3fff2724 40217a44  
3fff0960:  0000000f 3fff297c 3fff2724 401004d8  
3fff0970:  3fff09c0 3fff2524 3fff2724 00000000  
3fff0980:  00000000 3fff2524 3fff2724 01000000  
3fff0990:  3fff09c0 3fff2504 3fff2724 402177d0  
3fff09a0:  3fff1666 3fff2504 3fff1624 3fff2554  
3fff09b0:  00005d94 3fff2504 3fff1624 40205144  
3fff09c0:  3fff1624 00000000 3fff24cc 40204656  
3fff09d0:  3fff24cc 00003a98 30325443 3ffefa7c  
3fff09e0:  000022b3 3fff1624 3fff1666 3ffefa7c  
3fff09f0:  000022b3 3fff1624 3fff1666 402052a9  
3fff0a00:  3ffe9108 c2cb4634 3ffe9108 c2cb4634  
3fff0a10:  00000014 3fff1666 3fff1644 402038e8  
3fff0a20:  4010502c 0000000b 3fff1644 4020379e  
3fff0a30:  4023a69e 3ffef714 3ffef714 00000001  
3fff0a40:  4023b833 00036d1e 3ffef7dc 3ffefa7c  
3fff0a50:  00000014 0000000b 3ffef714 4020386c  
3fff0a60:  00000014 3ffefa7c 40205c2c 402027f0  
3fff0a70:  402015a6 3ffef714 3ffef714 402028b0  
3fff0a80:  feefeffe 3ffef714 3ffef9bc 402023ca  
3fff0a90:  3fffdad0 00000000 3ffefa74 40205c70  
3fff0aa0:  feefeffe feefeffe 3ffefa90 40100710  
<<<stack<<<

bogde, 2018-04-15:
see this: https://github.com/bogde/HX711/issues/73

Delirioo, 2018-04-16:
Hi, HX711 need to have separate vcc and vdd for working with esp8266 (wemos)?

Delirioo, 2018-04-16:
Another info: changing the wire diameter can cause read problem?
My problem is that i have the same read with load and without load.
I have connected the wire of the load cell to hx711 with 2mm cable

Delirioo, 2018-04-16:
[image]

I know it's not a solid connection, but it's just to test the operation quickly

I follow this https://cdn.instructables.com/F36/2JAR/J822Y1NX/F362JARJ822Y1NX.LARGE.jpg

and the sketch is the example of hx711 library

closed 2019-02-21 by bogde