bogde/HX711 #107: 4 load cells

originaldoctor, 2018-06-14:
Hi, 
Im using 4 load cells to  build a sensor that measures force and moment. My question is do i need to change something in HX711.h and HX711,.cpp to apply 4 loadcells?

thanks in advance

maurochiozzi, 2019-03-18:
If it's not a mistake, @originaldoctor ins't referring to the full bridge schematic. However, no matter what he is trying to achieve, he needs to explain better. For example, if it's four load cells with full-bridge built in strain gauges (like those https://cdn.sparkfun.com//assets/parts/1/0/6/2/9/13329-01.jpg) or they are load cells with single strain gauge like this one https://cdn.sparkfun.com//assets/parts/4/5/9/5/10245-01a.jpg, or it's just 4 strain gauges.

In the last two cases shown above, he will need to hookup the connection just as you showed.

amotl, 2019-03-18:
Hi there,

some people over at Hiveeyes are doing similar things, so you might enjoy reading [1].

With kind regards,
Andreas.

[1] https://community.hiveeyes.org/t/hx711-junction-board-fur-4-wagezellen/990

maurochiozzi, 2019-03-18:
Hey Andreas,

Thank you for the link, but since I don't speak german , I'll need some time at night to read it 😬 
But I have a question. I'm looking to use something quite similar:

I need to hookup 6 HX711 to get 6 different deformations, and have found this repo: [HX711-multi](https://github.com/compugician/HX711-multi) (forked from this repo), where @bogde helped in the development.

It's very clear on how to use: just plug all clock line together and for each data line, connect to a different pin on the Arduino. But on their example, they use the analog port for both clock and data wire. However, in this 'single' HX711 application they use digital pins connection. 

May you explain the difference/equivalences in using Analog/Digital ports for this application? They will work just as expected?

amotl, 2019-03-18:
Dear Mauro,

> May you explain the difference/equivalences in using Analog/Digital ports for this application? They will work just as expected?

As I am not a hardware guy in any way, I'm very sorry to disappoint you in that.

With kind regards,
Andreas.

scls19fr, 2019-04-16:
I'm also here because I want to be able to measure 4 masses independently at the same time. I also found @compugician repository https://github.com/compugician/HX711-multi which seems to support multiple load cells but which unfortunately haven't been updated since 2016. But I don't know if @bogde HX711 repository support. Some help will be great.

scls19fr, 2019-05-04:
Does this repository support multiple load cells or should I use https://github.com/compugician/HX711-multi which support multiple load cells but haven't been updated for a while? Kind regards

maurochiozzi, 2019-05-05:
@scls19fr I made up some tests using multiple load using this libs, and I had a quit success. For this simple test, I just hooked up every HX711 clock line in a single digital port and the data in separated digital ports.

For the @compugician repo, I haven't tested it since the @bogde worked ok.